openscad/openscad #1269: DXF export button?

whosawhatsis, 2015-03-18:
The new toolbar has a button for STL export, but of course that fails if the rendered geometry is 2D. Is there any way it could recognize that there is 2D geometry rendered and turn that button into a DXF export button?

You might also disable the button and make it greyed-out if the users hasn't done a render yet. Or even get really fancy and add a visual cue to indicate that there is rendered data, but the user has typed and/or previewed since then, so the render may not match what they're seeing.

t-paul, 2015-03-18:
I'm not sure that's the right way with more export options coming (e.g. why not SVG in case of 2D?). I think it would be partially solved by a generic export button that would come with #649.

In addition a way to configure the toolbar would be useful, unfortunately there seems to be no standard/automatic way with Qt to do that.

MichaelAtOz, 2015-03-18:
Perhaps a drop-down icon, if feasible, ie the button with a little down arrow on its side. Press the arrow and a drop-down with export types to select which you want as 'current' (ticked), click the button and the export is started. You could have items greyed out if they are not ready, like STL for a 2D render.

> Or even get really fancy and add a visual cue to indicate that there is rendered data, but the user has typed and/or previewed since then, so the render may not match what they're seeing.

There is still a problem (with MDI off) of loading another file (after F6'ing the previous one) then F5 (or auto F5) you see the new thing. But  the render of the previous one is still there, choose wireframe and you see the old render. Press export and you export the old thing.

t-paul, 2022-10-07:
Toolbar now has a simple solution to configure (via Preferences) one 3d and one 2d export button. Hopefully at some point we get a fully customizable toolbar, but this at least helps users who don't have to switch between export formats very often.

closed 2022-10-07 by t-paul