openscad/openscad #1271: Geometry glitch in openscad 2015.03

tbuser, 2015-03-20:
I found this customizer script http://www.thingiverse.com/thing:677959

It used to create a solid heart, but now the heart section has a gap down the center.  You can see it clearly here: http://www.thingiverse.com/thing:731628 vs. an old one here: http://www.thingiverse.com/thing:724620

kintel, 2015-03-20:
Thanks for the test. The reason this fails is that we're a bit more aggressive when merging close vertices in 2015.03 than earlier, and apparently we sometimes fail to handle that properly in all cases.
The reason this particular design triggers this is that it's designed really small; the small heart has a size of 0.2mm.

kintel, 2024-03-25:
Closing as this has since been fixed.

closed 2024-03-25 by kintel