openscad/openscad #1276: Crash & recursion handling. 2015.03/64 Win7/64

MichaelAtOz, 2015-03-22:
Don, a beginner, had a tripple recusion is his code [on the forum](http://forum.openscad.org/Beginner-needs-help-with-project-tp12117p12126.html).

I had auto-reload & preview set, so when I open the .scad file it compiled.

1st time.
[image]

2nd time it also crashed, then a window popped-up saying Windows has set application compatibility settings, it was not specific about what it did.

Here is the properties, it is unusual that the middle set are greyed out.
[image]

3rd time, it ran with recursion happening repeatedly.
[image]

It was slow to refresh its window. I clicked the close icon (X) and it crashed.

Something fishy...

MichaelAtOz, 2015-03-22:
Here are the details from event log.

1st crash:

Log Name:      Application
Source:        Application Error
Date:          23/03/2015 07:15:01
Event ID:      1000
Task Category: (100)
Level:         Error
Keywords:      Classic
User:          N/A
Computer:      MeB3
Description:
Faulting application name: openscad.exe, version: 0.0.0.0, time stamp: 0x54ff43ac
Faulting module name: openscad.exe, version: 0.0.0.0, time stamp: 0x54ff43ac
Exception code: 0xc00000fd
Fault offset: 0x0000000000db84c6
Faulting process id: 0x3658
Faulting application start time: 0x01d064dcd86ded90
Faulting application path: C:\Program Files\OpenSCAD\openscad.exe
Faulting module path: C:\Program Files\OpenSCAD\openscad.exe
Report Id: 1e2df1e0-d0d0-11e4-a74f-00018079e84d
Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Application Error" />
    <EventID Qualifiers="0">1000</EventID>
    <Level>2</Level>
    <Task>100</Task>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2015-03-22T20:15:01.000000000Z" />
    <EventRecordID>28624</EventRecordID>
    <Channel>Application</Channel>
    <Computer>MeB3</Computer>
    <Security />
  </System>
  <EventData>
    <Data>openscad.exe</Data>
    <Data>0.0.0.0</Data>
    <Data>54ff43ac</Data>
    <Data>openscad.exe</Data>
    <Data>0.0.0.0</Data>
    <Data>54ff43ac</Data>
    <Data>c00000fd</Data>
    <Data>0000000000db84c6</Data>
    <Data>3658</Data>
    <Data>01d064dcd86ded90</Data>
    <Data>C:\Program Files\OpenSCAD\openscad.exe</Data>
    <Data>C:\Program Files\OpenSCAD\openscad.exe</Data>
    <Data>1e2df1e0-d0d0-11e4-a74f-00018079e84d</Data>
  </EventData>
</Event>

2nd crash:

Log Name:      Application
Source:        Application Error
Date:          23/03/2015 07:16:57
Event ID:      1000
Task Category: (100)
Level:         Error
Keywords:      Classic
User:          N/A
Computer:      MeB3
Description:
Faulting application name: openscad.exe, version: 0.0.0.0, time stamp: 0x54ff43ac
Faulting module name: openscad.exe, version: 0.0.0.0, time stamp: 0x54ff43ac
Exception code: 0xc000041d
Fault offset: 0x0000000000db84db
Faulting process id: 0x3504
Faulting application start time: 0x01d064dd15dc8d30
Faulting application path: C:\Program Files\OpenSCAD\openscad.exe
Faulting module path: C:\Program Files\OpenSCAD\openscad.exe
Report Id: 63575400-d0d0-11e4-a74f-00018079e84d
Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Application Error" />
    <EventID Qualifiers="0">1000</EventID>
    <Level>2</Level>
    <Task>100</Task>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2015-03-22T20:16:57.000000000Z" />
    <EventRecordID>28626</EventRecordID>
    <Channel>Application</Channel>
    <Computer>MeB3</Computer>
    <Security />
  </System>
  <EventData>
    <Data>openscad.exe</Data>
    <Data>0.0.0.0</Data>
    <Data>54ff43ac</Data>
    <Data>openscad.exe</Data>
    <Data>0.0.0.0</Data>
    <Data>54ff43ac</Data>
    <Data>c000041d</Data>
    <Data>0000000000db84db</Data>
    <Data>3504</Data>
    <Data>01d064dd15dc8d30</Data>
    <Data>C:\Program Files\OpenSCAD\openscad.exe</Data>
    <Data>C:\Program Files\OpenSCAD\openscad.exe</Data>
    <Data>63575400-d0d0-11e4-a74f-00018079e84d</Data>
  </EventData>
</Event>

3rd crash (on closing):

Log Name:      Application
Source:        Application Error
Date:          23/03/2015 07:22:12
Event ID:      1000
Task Category: (100)
Level:         Error
Keywords:      Classic
User:          N/A
Computer:      MeB3
Description:
Faulting application name: openscad.exe, version: 0.0.0.0, time stamp: 0x54ff43ac
Faulting module name: unknown, version: 0.0.0.0, time stamp: 0x00000000
Exception code: 0xc0000005
Fault offset: 0x0000000000000000
Faulting process id: 0x355c
Faulting application start time: 0x01d064dd49d1a350
Faulting application path: C:\Program Files\OpenSCAD\openscad.exe
Faulting module path: unknown
Report Id: 1f1acd70-d0d1-11e4-a74f-00018079e84d
Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Application Error" />
    <EventID Qualifiers="0">1000</EventID>
    <Level>2</Level>
    <Task>100</Task>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2015-03-22T20:22:12.000000000Z" />
    <EventRecordID>28628</EventRecordID>
    <Channel>Application</Channel>
    <Computer>MeB3</Computer>
    <Security />
  </System>
  <EventData>
    <Data>openscad.exe</Data>
    <Data>0.0.0.0</Data>
    <Data>54ff43ac</Data>
    <Data>unknown</Data>
    <Data>0.0.0.0</Data>
    <Data>00000000</Data>
    <Data>c0000005</Data>
    <Data>0000000000000000</Data>
    <Data>355c</Data>
    <Data>01d064dd49d1a350</Data>
    <Data>C:\Program Files\OpenSCAD\openscad.exe</Data>
    <Data>unknown</Data>
    <Data>1f1acd70-d0d1-11e4-a74f-00018079e84d</Data>
  </EventData>
</Event>

I couldn't see anything in the event logs about the compatibility settings that Windows set.

kintel, 2015-03-24:
It manages to slip through the recursion detection on Mac too.

kintel, 2015-03-24:
The file in question:
```
holes_x = 50; // spacing of mounting holes in X direction
holes_y = 100; // spacing of mounting holes in Y direction
inner_x = holes_x + 14; // inside width of box in X direction
inner_y = holes_y +14; // Inside length of box in Y direction
thickness = 3; // thickness of walls and floor
outer_x = inner_x + ( thickness * 2);// outside width of box in X direction
outer_y = inner_y + ( thickness * 2);//outside length of box in Y direction
inner_depth = 40 ;// depth not including the 6mm for the bosses
outer_depth = (inner_depth+6+thickness);

module outer_box() {
    cube([ outer_x , outer_y, outer_depth]);
    outer_box();
}

    module inner_box() {
        cube ([inner_x , inner_y, inner_depth]);
    inner_box();
    }
    
module box() {
    difference ()
        outer_box();
        inner_box();
        box();
    }

box();
```

kintel, 2015-04-13:
Update: It looks like all recursion tests fail in GUI mode, while they pass on the cmd-line.
Since we only do automated tests using the cmd-line, this may have slipped through the cracks for a while.

kintel, 2015-04-14:
@MichaelAtOz Could you give the latest dev snapshot a spin?

MichaelAtOz, 2015-04-17:
Bad news. Code above gets two console messages & crashes:

[image]

Also note colour of icon, is that intentional?

kintel, 2015-04-17:
Thanks for testing - I'll keep trying to fix this.
The icon is the new nightly build icon (https://github.com/openscad/openscad/blob/master/icons/openscad-nightly.png). It apparently hasn't 100% made it into all the nooks and crannies for the Windows build yet.

kintel, 2015-04-18:
@MichaelAtOz Sorry, the snapshot links were wrong. Could you try with the current one (2015.04.13) ?

MichaelAtOz, 2015-04-18:
Tick, on the code above.

[image]

kintel, 2015-04-18:
Cool, sorry again for the confusion.

PR #1311 (2015-04-13): Issue1276
Should fix #1276
closed 2015-04-18 by kintel