openscad/openscad #1277: Automatic reload resets to wrong view

farbo, 2015-03-22:
On version 2015.03.22, the automatic reload will change my view on update. The view will change to whatever it was on the latest manual reload. This happens only on a specific, complex assembly, and has been going on for at least about 2 months (as long as I've been using the git version). Small parts work fine, and if I set up a new copy of the whole assembly it works fine too.

This is the project: https://github.com/farbo/bengraver (scad/frame_assembly.scad)

t-paul, 2015-03-22:
Reproducible (with master / Linux) when touching one of the `use`d files (e.g. `touch bar.scad`).

closed 2015-03-23 by kintel

kintel, 2015-04-14:
@farbo The latest dev snapshot has a fix for this.

PR #1279 (2015-03-23): Update temporal variables for each evaluation, not for each compilation....
... This ensures that camera parameters are correct even if we reuse an already compiled document. Fixes #1277
commit 5ad4b130 (2015-03-23): Update temporal variables for each evaluation, not for each compilation. This ensures that camera parameters are correct even if we reuse an already compiled document. Fixes #1277