openscad/openscad #1282: rotate_extrude() of 2D object in the negative X half plane results in an inside-out object

kintel, 2015-03-24:
While this is not a common workflow, people tend to get into this situation. This should be easy enough to fix. Testing needed.

TakeItAndRun, 2015-03-24:
There are two solutions:
1) ignore everything less than x<0
2) reflect the negative half plane into the positive half plane and rotate_extrude
I'd prefer 2).

kintel, 2015-03-24:
I think 2) makes sense. Of course keeping the current limitation that objects crossing the Y axis are not going to make OpenSCAD happy.

atartanian, 2015-04-29:
i can think of another solution, which would be to clip the polygon along the y axis, flip it across the y axis and then union it with the part of the shape that is on the +x side. this results in the same shape you get if you have an object crossing the y axis, but without the inverted faces

atartanian, 2015-04-29:
which is basically #2 but with cleaning up the polygon before rotating it

thehans, 2015-11-21:
This is handled in the recently merged PR #1478 and can be closed.

closed 2015-11-21 by t-paul