openscad/openscad #1284: Animation Flicker

davidthings, 2015-03-28:
Since the last major release (or soon before it) I'm pretty sure Animation has changed for the worse. 

Currently on Mac and Linux (nightly and release) animation flickers quite considerably more than it used to.  I don't think that it's running any slower, I think, perhaps, the screen is being cleared a bit too early? 

I just went back to version 2013.01+dfsg-2.2 and although most of my code didn't  run, (children() spoil everything) what did show up was smooth and flicker free.  Back on the current release, showing only that which displayed on the old version, the display still flickered. 

It's a bummer, because animation was rocking my world. 

I have some pretty crappy video that shows the problem on my machine. 

https://www.dropbox.com/s/47yvwiv123408z9/os_flicker.MOV?dl=0

The video starts and ends at full speed with a slo-mo section in the middle.  I hope this helps.  I'd really like flicker free animation back!

TakeItAndRun, 2015-03-28:
On Windows with OpenSCAD version 2015.03 it flickers, too.

closed 2015-04-13 by kintel

kintel, 2015-04-14:
@davidthings The latest dev snapshot has a fix for this. Could you give it a spin?

davidthings, 2015-04-14:
Already tried!

At least on the Mac, works like a charm.

Thank you SO much.

PR #1310 (2015-04-13): Issue1284
This should fix #1284.