openscad/openscad #1286: import dxf does not work with non-ascii file names

1i7, 2015-03-29:
Hello, Windows 8, OpenSCAD 1015.03

import fails to load dxf files with non-ascii file name.

this will load dxf file to workspace:
import(file = "C:/gear/gear.dxf");

this will not load same dxf file with different name to workspace:
import(file = "C:/gear/колесо.dxf");

this seems to be regression, at least in 2013.06 running on the same system same file loads fine.

kintel, 2015-04-02:
This is a problem in multiple places. Looks like some refactoring gone wrong.
I'll look into it.
Suggestion:
* All filenames should be utf-8 std::string or QString internally
* Conversions between std::string and boost::filesystem::path should use an utf-8 codec
* Any usage of fopen and iostreams are broken by design under Windows, and we need to use wide char variants (or a wrapper library)
* We need manual and automated tests for all code paths (GUI and file operations)

kintel, 2015-04-02:
We'll handle this in #1296