openscad/openscad #1288: Openscad 2015.03 (linux) fails to export png

gratefulfrog, 2015-03-30:
When running on linux Ubuntu 32bit, I try to export a png, but it captures the screen including the dialog where I choose the filename for the png file.  see attached png resulting from File/Export/Export as image...

This worked fine in the previous version.

[image]

ehooo, 2015-04-01:
Also for [other Linux](http://www.openscad.org/downloads.html#linux) now only have available version

kintel, 2015-04-01:
This functionality is currently dependent on the GPU, opengl drivers and window system; we simply read the framebuffer and save it to file. This isn't always safe as each platform manages framebuffer resources differently.
We should probably rewrite this to re-render the current scene into a separate buffer before exporting.

gratefulfrog, 2015-04-01:
Yes,the current release really fails on linux. Althought, the release
available via the Ubuntu package mgr exports images correctly.

IMHO, this bug was introduced in a recent release.

Thanks for your help!
B

kintel, 2015-04-01:
The old version exported the second-to-last image, which caused the wrong image to be exported in some cases. The new version exports the correct image, but breaks on platforms where each application windows isn't backed by separate framebuffer resources. To my knowledge this means it's likely to fail on certain X11-based systems.

kintel, 2015-04-13:
@gratefulfrog Could you tell us more about your system? e.g. Help->Library Info ?
We've tested this on multiple Linux installations without being able to reproduce..

gratefulfrog, 2015-04-13:
@kintel 

This is my "library Info" : 

OpenSCAD Version: 2015.03
System information: Linux 3.2.0-80-generic #116-Ubuntu SMP Mon Mar 23 17:11:32 UTC 2015 i686 Ubuntu precise (12.04.5 LTS) 4 CPUs 1.72 GB RAM
Compiler, build date: GCC "4.6.3" 32bit, Mar 24 2015
Boost version: 1_46_1
Eigen version: 3.0.5
CGAL version, kernels: 3.9, N4CGAL9CartesianINS_4GmpqEEE, N4CGAL18Extended_cartesianINS_4GmpqEEE, N4CGAL5EpeckE
OpenCSG version: OpenCSG 1.3.2
Qt version: 4.8.1
QScintilla version: 2.6.1
MingW build: No
GLib version: 2.32.4
Application Path: /usr/bin
Documents Path: /home/bob/.local/share
Resource Path: /usr/share/openscad
User Library Path: /home/bob/.local/share/OpenSCAD/libraries
User Config Path: /home/bob/.config/OpenSCAD
Backup Path: /home/bob/.local/share/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
/home/bob/.local/share/OpenSCAD/libraries
/usr/share/openscad/libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
GLEW version: 1.6.0
OpenGL Version: 3.3.0 NVIDIA 304.125
GL Renderer: ION/integrated/SSE2
GL Vendor: NVIDIA Corporation
RGBA(8880), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Using QGLWidget

GL Extensions:
GL_ARB_ES2_compatibility
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_explicit_attrib_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_imaging
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
[... 191 lines cut ...]
GL_OES_depth_texture
GL_OES_element_index_uint
GL_OES_fbo_render_mipmap
GL_OES_get_program_binary
GL_OES_mapbuffer
GL_OES_packed_depth_stencil
GL_OES_point_size_array
GL_OES_point_sprite
GL_OES_read_format
GL_OES_rgb8_rgba8
GL_OES_standard_derivatives
GL_OES_texture_3D
GL_OES_texture_float
GL_OES_texture_float_linear
GL_OES_texture_half_float
GL_OES_texture_half_float_linear
GL_OES_texture_npot
GL_OES_vertex_array_object
GL_OES_vertex_half_float
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum

kintel, 2015-04-16:
@gratefulfrog Did you build your own binaries? We cannot reproduce this, but have some ideas which you could try out. If you'd be willing to help, the easiest would be to find me (kintel) on IRC (#openscad on Freenode), and I can provide you with some patches.

kintel, 2015-04-16:
@gratefulfrog As a start, could you try building the issue1288 branch? https://github.com/openscad/openscad/tree/issue1288

gratefulfrog, 2015-04-17:
Hello,

I did not build from source I used a .deb from ppa:openscad/releases
installed with apt-get.

I am happy to help, but am away from my computer for 10 days. I will try to
your suggestions when I get back!
Thanks for the response!
Bob

gratefulfrog, 2015-04-17:
I experienced the same thing again today. Export image captures the screen
including the file name dialog box. This is really a problem...

I will attempt to build from source when I get back in 10 days.

Thanks,
Bob

closed 2015-04-17 by kintel

gratefulfrog, 2015-04-17:
> Closed #1288 <https://github.com/openscad/openscad/issues/1288> via
> a1bece5

Closed? Really? Is there a patch I can install?

kintel, 2015-04-17:
@gratefulfrog The nightly build will have this - sometime tonight.

gratefulfrog, 2015-04-17:
Thanks!  Sorry to be a noobie, but how to I get the nightly build? Is there
a .deb or  a repo to add to apt-get?

Thanks for all your help,
Bob

kintel, 2015-04-17:
See http://www.openscad.org/downloads.html#snapshots

gratefulfrog, 2015-04-18:
Hi,
Thanks I followed those instructions but I am running Ubuntu 12.04 LTS...
and nothing appears in my package manager?

is there repo I can add with apt-get and a package I can fetch?  Any help
would be greatly appreciated.

Cheers,
B

kintel, 2015-04-18:
@gratefulfrog Oh, 12.04 - sorry, we don't support that version for the snapshot builds.

commit a1bece5c (2015-04-17): Merge pull request #1316 from openscad/issue1288

Grab framebuffer before opening file dialog. Should fix #1288