openscad/openscad #1297: Extremely slow performance

ntoff, 2015-04-03:
2015.03 is atrociously slow. Typing in the editor lags and the preview window, with its constant updating, is jerky. Also while working with files on a network drive, OpenSCAD is constantly streaming data, it never stops.

PC is an Intel i7 2700k with a GTX760 and 16gigs of ram, it's not new but it's certainly not a crap PC.

nophead, 2015-04-03:
Try turning off auto reload. If you have lots of included files it will be
checking their times and that could be slow over a network.

ntoff, 2015-04-03:
Yeah I just found it. I searched all through the preferences but nothing, turns out it's in the "Design" menu.

t-paul, 2015-04-04:
As there are no more details that could indicate a specific problem... closing as duplicate of #1237.
(Please subscribe to that issue if you are interested in progress notifications).

closed 2015-04-04 by t-paul