openscad/openscad #1304: Export models for KiCad

cb88, 2015-04-07:
I would love to use OpenSCAD for implementing KiCad part  models... however it is quite error prone to export as STL then touch up in another tool then export as VRML (.wrl) to KiCad.

It would be ideal if it were possible to export to KiCad models with color information.

This has been mentioned on the mailing list before but there didn't seem to be an issue tracking it.

kintel, 2015-04-07:
..so in short, you'd like to see VRML export resembling OpenSCAD previews?
Are there any particular requirement or limitations in KiCad's VRML importer?

t-paul, 2015-04-07:
I found a number of discussions about problematic VRML files, but not much details. It seems the best supported generator is Wings3D.

Some examples: https://github.com/cwoodall/scad-parts-library (http://happyrobotlabs.com/posts/tutorials/tutorial-3d-kicad-parts-using-openscad-and-wings3d/)

cb88, 2015-04-08:
From what I understand KiCad can handle VRML 2 only (with some recent improvements so it might be good to talk to them about it while it is fresh on their minds) ... but you have to make sure you give it clean output which means passing it through a tool that knows how to clean that up like wings3d (some cad tools duplicate geometry etc...)

Also, I tried exporting from OpenSCAD importing in FreeCad and then exporting as VRML to KiCad but something was lost in the translation as groups/unions were no longer correctly aligned and two subtracted cubes were no longer being subtracted but added to my model.

TakeItAndRun, 2015-04-08:
I had a look at the KiCad-files here:
https://github.com/cwoodall/scad-parts-library
To me it looks as they are using a format that is a cross between .stl
(listing vertex-points for every triangular face,together with the normal
vector on this face) and OpenSCADs polyhedron command:
It lists
1) all the vertex-points
2) all the faces like faces of the polyhedron command (followed by a -1 for
every face)
3) all normal vectors to all the faces
4) again all the faces like faces of the polyhedron command (followed by a
-1 for every face)

Now OpenSCAD does not output the normal vectors on .stl output, but, as I
understand it, the are calculated and kept internally.

So all the information is there to output the geometry as IndexedFaceSet of
the VRML format.

-- 
stempeldergeschichte@googlemail.com <karsten@rohrbach.de>

P.S. Falls meine E-Mail kürzer ausfällt als Dir angenehm ist:
Ich probiere gerade aus kurze Antworten statt gar keine Antworten zu
schreiben.
Wenn Du gerne mehr lesen möchtest, dann lass es mich bitte wissen.

P.S. In case my e-mail is shorter than you enjoy:
I am currently trying short replies instead of no replies at all.
Please let me know, if you like to read more.

Enjoy!

TakeItAndRun, 2015-04-08:
My guess to why the import of OpenSCAD - .stl -files to wing3D fails is,
that OpenSCAD sets all the  facet normals to  0 0 0.

2015-04-08 16:03 GMT+02:00 Peter Falke <stempeldergeschichte@googlemail.com>
:

> I had a look at the KiCad-files here:
> https://github.com/cwoodall/scad-parts-library
> To me it looks as they are using a format that is a cross between .stl
> (listing vertex-points for every triangular face,together with the normal
> vector on this face) and OpenSCADs polyhedron command:
> It lists
> 1) all the vertex-points
> 2) all the faces like faces of the polyhedron command (followed by a -1
> for every face)
> 3) all normal vectors to all the faces
> 4) again all the faces like faces of the polyhedron command (followed by a
> -1 for every face)
>
> Now OpenSCAD does not output the normal vectors on .stl output, but, as I
> understand it, the are calculated and kept internally.
>
> So all the information is there to output the geometry as IndexedFaceSet
> of the VRML format.
>
>
>
>
>
>> From what I understand KiCad can handle VRML 2 only (with some recent
>> improvements so it might be good to talk to them about it while it is fresh
>> on their minds) ... but you have to make sure you give it clean output
>> which means passing it through a tool that knows how to clean that up like
>> wings3d (some cad tools duplicate geometry etc...)
>>
>> Also, I tried exporting from OpenSCAD importing in FreeCad and then
>> exporting as VRML to KiCad but something was lost in the translation as
>> groups/unions were no longer correctly aligned and two subtracted cubes
>> were no longer being subtracted but added to my model.
>>
>>
>
>
>
> stempeldergeschichte@googlemail.com <karsten@rohrbach.de>
>
> P.S. Falls meine E-Mail kürzer ausfällt als Dir angenehm ist:
> Ich probiere gerade aus kurze Antworten statt gar keine Antworten zu
> schreiben.
> Wenn Du gerne mehr lesen möchtest, dann lass es mich bitte wissen.
>
> P.S. In case my e-mail is shorter than you enjoy:
> I am currently trying short replies instead of no replies at all.
> Please let me know, if you like to read more.
>
> Enjoy!

-- 
stempeldergeschichte@googlemail.com <karsten@rohrbach.de>

P.S. Falls meine E-Mail kürzer ausfällt als Dir angenehm ist:
Ich probiere gerade aus kurze Antworten statt gar keine Antworten zu
schreiben.
Wenn Du gerne mehr lesen möchtest, dann lass es mich bitte wissen.

P.S. In case my e-mail is shorter than you enjoy:
I am currently trying short replies instead of no replies at all.
Please let me know, if you like to read more.

Enjoy!

nophead, 2015-04-08:
That seems to be a new bug because OpenScad used to output facet normals that were not all 0. Wings still crashes on those though. It gives a crash log and the relevant line seems to be this:
             {'EXIT',{{badmatch,<<"CAD_Model\r\n">>},

It looks like it chokes on the first line, which is: solid OpenSCAD_Model

nophead, 2015-04-08:
It seems to be simply that Wings only accepts binary STL files. See http://wings3d.com/forum/showthread.php?tid=510

If I convert to binary with NetFabb studio it imports correctly into Wings. 

For component models STL is not very good though as it loses the colours.

cb88, 2015-04-10:
Also, My request with this bug report was to add support to export directly from KiCad (Probably VRMLv2)... since thats less confusing and less prone to error once it is working. And once it works I would imagine the KiCad guys would test with it as well to make sure it continues working since it would be highly preferable to the current workflow.

kintel, 2015-04-11:
The challenge is that color information isn't preserved across CSG operation boundaries, and VRML doesn't include feature for describing CSG operations. ..so we need a better post-evaluation representation (which would benefit other potential export formats as well, e.g. OBJ).

cb88, 2015-04-23:
AMF lookes like it might be a good format for kicad and openscad to support as well. I assume the AMF support in OpenSCAD now doesn't export the color/material either.

Seems this has been talked about before as well... http://forum.openscad.org/Feature-request-STL-2-0-aka-AMF-export-amp-import-support-td1111.html

cb88, 2015-04-23:
It would appear that KiCAD can render X3D as well... which seems to be the successor of VRML.

I was able to import an STL from OpenSCAD into blender and export as X3D the part renders well except for no material colors except for the base color (which appears to be a KiCAD bug as meshlab renders it 100% correctly)

bobc, 2015-05-12:
I inquired about this before so +1

Kicad is tuned to read files generated by Wings3d, and even then only supports a subset, though this support has improved. Additionally, it uses "deci-inch" as the size unit.
To export to Kicad, I wrote a tool to read STL, convert the units, and export indexed face set in the style Kicad likes. This works fine for monochrome, to support color, my tools imports several STLs, the user assign a color for each one, and the tools output combined file.

If OpenSCad support color and AMF, it would be an easy translation. As regards CSG, for VRML it is display only, so only surface properties are relevant. In that case, the CSG ops are simple. Additionally, AMF requires non-overlapping volumes. So the way I see color working in OpenSCAD, is to create multiple top level volumes, then a single color (material) property is derived for each volume.

As a use case, I have a typical PCB connector, metal pins in a plastic body.

bobc, 2015-05-12:
Thinking about a *desired* syntax, for discussion:

    export ("connector.amf")
	{
	    material (id="1",name="body",color=[0.1,0.1,0.1]);
	    material (id="2",name="pin",color=[0.75,0.75,0.75]);
	
	    //AMF: object id=1  
	    //VRML: object1
	    object (id="1")
	        volume (materialid="1")
	            connector_base();
	
	    //AMF: object id=2
	    //VRML: object2
	    object (id="2")
	        volume (materialid="2")
	            connector_pins();
	}

A default / backward compatible behaviour : any top level objects that are not "export" are handled as now. 
export () creates a tree of nodes which can be processed with or without the color attribute to create preview or render
A new menu item "Export" uses the filename and implied format in the export() statements.
"Export as" maps the tree to desired output format? ...that might not work.

t-paul, 2015-05-12:
Defining magic module names could cause trouble later on. It is probably better to attach this information as meta data to the normal tree and leave the actual interpretation to the specific export (which would need some extension too - #649).

Still, the main challenge is to get the CSG operations to support colors/materials. Right now the color information is only there in preview mode which is not enough to export AMF or VRML, I think.

bobc, 2015-05-13:
>  It is probably better to attach this information as meta data to the normal tree and leave the actual interpretation to the specific export

That is exactly what I was trying to do - can you suggest how that should be done?

> the main challenge is to get the CSG operations to support colors/materials

I think that challenge is extremely hard, both semantically and implementation wise. So hard that I suggest avoiding it. In the case of union, what properties should the intersection have? Remember we want to support material properties other than color. We would need a blending function to specify the properties of the intersection. Even providing a sensible default is problematic. 

Currently, I believe a union creates a new single mesh. With blending, we would need to create multiple new meshes, corresponding to properties of A, B, or mix of A+B. In order to export these, each mesh needs to be exported as a separate object, because in VRML color properties are assigned per object. AFAIK, this requires a new CSG engine.

Do we want to wait until we implement a new CSG engine? My fear is that if we do, we will never get round it.

Therefore, I think that CSG ops should not take account of properties. The user should apply properties to a top level object. This KISS approach puts control in the hands of the user, rather than doing things automagically, but achieves the goal of a useful and usable way to create objects with color/material properties, which is a lot better than nothing.

kwendenarmo, 2015-05-23:
+1 to export option for KiCad, but:

-   VRMLv2 is obsolete, and KiCad implementation only supports a part of the specifications.  
-   X3D support is active.  
-   The KiCad 3d-viewer roadmap is in way to support STEP files instead of above mentioned.

References:

-   [3d-viewer future discussion](https://lists.launchpad.net/kicad-developers/msg17687.html)  
-   [FreeCAD 0.15](https://lists.launchpad.net/kicad-developers/msg17929.html)  
-   [3d-viewer redesign proposal](https://lists.launchpad.net/kicad-developers/msg17868.html)

So, Would not it be logical that both applications were common meeting point?

t-paul, 2015-05-23:
Hmm, adding STEP is likely a huge project, even with the existing stepcode library. The IGES lib looks promising https://github.com/cbernardo/libIGES though.

kwendenarmo, 2015-05-23:
They are also working for IGES support, yes

bobc, 2015-07-29:
Well, since saying "this is too difficult", I have to do some eating of words, and hats if I had one. The past 2 months I have been delving into the code, and not only have I added a new CSG engine using the Carve library, I have also implemented rendering of color.

There are still quite a few loose ends, functions such as hull and minkowski are not implemented with Carve, but I am quite pleased at the progress with respect to my original goals.

I am still some way from a beta release, but the source is at https://github.com/bobc/openscad/tree/carve_csg if anyone is interested.

cb88, 2015-07-30:
@Bob Thanks for all the hard work! I'm glad to see this support being added... it is much needed.

easyw, 2015-08-11:
Hi to all, 
please have a look there
https://forum.kicad.info/t/kicad-stepup-new-exporter-for-3d-mcad-feedbacks-are-welcome/1048
I've done a script to add STEP AP214 support to kicad
From a STEP model it is possible to export with FreeCAD to vrlm with a format fully supported by kicad new release
Then scripting 3D models to STEP and export for kicad should be supported now ...

shufps, 2017-01-16:
> It would be ideal if it were possible to export to KiCad models with color information.

This is indeed possible ... Export to CSG in openscad, start freecad, import CSG, export WRL, import WRL in kicad, all colors are here and everything is fine :)

Muriukidavid, 2017-01-20:
Nope, tried, that does not work. CSG from OpenSCAD causes errors on importing to FreeCAD

av-p, 2017-04-08:
Thank you, shufps. Your method works perfectly. Tested on a few models, freecad version is 0.16.
I've found a few rules to be followed to avoid freecad CSG-import problems.
Model should be drawn with a color setting in a root of structure tree, and scale setting in a next level, like this:

ScaleFactor = 1/2.54; // Scaling is required to import a model into KiCAD with correct size

color ("DimGray")
	scale ( [ScaleFactor,ScaleFactor,ScaleFactor] )
		{
		// Draw a chip case with pins cutouts here
		}

color ("Gold")
	scale ( [ScaleFactor,ScaleFactor,ScaleFactor] )
		{
		// Draw pins here
		}

If this strucure is not followed, you will have to restore colors manually in most cases. Looks like, freecad is not able to correctly propogate color settings through model structure tree.

KeithSloan, 2017-07-09:
If you have any problems going from OpenSCAD to KiCAD via FreeCAD please report problem files in the FreeCAD forum https://forum.freecadweb.org/viewforum.php?f=3. Problems will only get fixed if you report them and we have test cases.

tcurdt, 2017-07-09:
If I understand the original intention of this issues it is about exactly **not** going through any other tools.

cb88, 2017-07-11:
Exactly, faffing about importing into this tool and that should not be considered normal... and was exactly why I originally posted this feature request. While it may be a technical challenge OpenSCAD should be able to export models with color directly. The other use case it is important for is multi extruder printing I think AMF is usually desirable there. If you want to continue posting how to export from openscad using other tools it seems your own blog etc... would be more fitting than this topic as it just isn't what it is about.

KeithSloan, 2017-07-11:
"faffing about importing into this tool and that should not be considered normal."
Well with FreeCAD you can open a SCAD file and export as vrml or STEP file. I would not call it faffing about, you are just using FreeCAD instead of OpenSCAD

podsvirov, 2021-09-18:
Looks like this issue already resolved via #3883 and can be closed.

t-paul, 2021-09-18:
Thanks @podsvirov, I think we can close this issue. The current solution, available in the dev snapshots, is limited regarding coloring (single fixed OpenSCAD yellow color) but this probably should be tracked by a separate dedicated issue as it's not specifically related to VRML. There might be a short-cut allowing reasonable multi-color output, full color support is probably a bit further out (but would certainly be awesome).

closed 2021-09-18 by t-paul