openscad/openscad #1308: Compile Error in Ubuntu 14.04

ankush1995, 2015-04-12:
I tried to build the openscad environment on Ubuntu (14.04) , when I ran the command
qmake openscad.pro
I got the following ---->
Project MESSAGE: If you're building a development binary, consider adding CONFIG+=experimental
WARNING: /home/ankush/openscad/scintilla.pri:12: Unable to find file for inclusion qscintilla2
Project MESSAGE: Using local copy of qscintilla2.prf instead.

ankush1995, 2015-04-12:
Also on running make install command , getting the same message

t-paul, 2015-04-12:
That's no error, just a warning. The Ubuntu-14.04 dev package of qscintilla2 does not ship the prf file in the correct place. The warning just explains we are trying to work around this issue.

closed 2015-04-12 by t-paul

ankush1995, 2015-04-12:
But I am not able to make the Make file and also not able to run the application.

t-paul, 2015-04-12:
Then there must be some additional error, the message above is normal and not causes the build to fail.

ankush1995, 2015-04-12:
After running Make I got----->

/usr/lib/x86_64-linux-gnu/qt4/bin/uic src/MainWindow.ui -o objects/ui_MainWindow.h -tr _
make: /usr/lib/x86_64-linux-gnu/qt4/bin/uic: Command not found
make: *** [objects/ui_MainWindow.h] Error 127

t-paul, 2015-04-12:
This looks like the `libqt4-dev` package is not correctly installed, this should normally install the dependency `libqt4-dev-bin` which has the `uic` program.

ankush1995, 2015-04-13:
I am geting the following error on running make command :
In file included from src/scintillaeditor.cpp:6:0:

src/scintillaeditor.h:8:32: fatal error: Qsci/qsciscintilla.h: No such file or directory
 #include <Qsci/qsciscintilla.h>
                                ^
compilation terminated.
make: *** [objects/scintillaeditor.o] Error 1

It seems that I dont have Qsciscintilla on my system. Can u tell me the appropriate linux package for that?

t-paul, 2015-04-14:
That's `libqscintilla2-dev` (or `libqt5scintilla2-dev` when using Qt5).

alankruppa, 2015-06-03:
Had the same error when building version 2015-3. Installing libqscintilla2-dev worked for me. Thanks!

FluxIX, 2015-07-02:
I am trying to build from source and have installed libqscintilla2-dev (I am using Qt 5.2.x), but I still get the following error when I try to compile:
> /home/chris/openscad/src/scintillaeditor.h:8: error: Qsci/qsciscintilla.h: No such file or directory
> #include <Qsci/qsciscintilla.h>
>                                ^

All the dependencies checked by the check-dependencies.sh script have passed.

Any idea what I need to do to resolve this?

alankruppa, 2015-07-02:
You might try installing libqt5scintilla2-dev for Qt5 mentioned above by t-paul.

FluxIX, 2015-07-02:
My error. I had meant to say I have libqt5scinitilla2-dev installed and I was seeing the compiler error.

alankruppa, 2015-07-02:
Gotcha. Did you try making with qmake-qt4?

FluxIX, 2015-07-02:
Same error at the same location.

alankruppa, 2015-07-02:
That's all I've got. Good luck with it.