openscad/openscad #1309: Windows: Crash when accessing symlinked OpenSCAD file

kintel, 2015-04-13:
Attempting to preview or render a file in a symlinked directory causes the process to crash:

    linktest/
        main.scad
        slave.scad

main.scad:

     include <slave.scad>

slave.scad:

     sphere(10);

In a DOS promt, create a symlink:

    $ mklink /d linked linktest

Coolway99, 2016-06-05:
What I thought was a new issue turned out to be a duplicate of this. My entire system is heavily symlinked together, therefore simply having OpenSCAD zip on the desktop crashes it upon boot, let alone including two files _in the same directory_ together.

Can confirm it using Windows 7, OpenSCAD versions 2015.03, 2016.05.12. 

This is the only application I have so far that just doesn't like symlinks.

Coolway99, 2016-06-05:
A bit of digging points this to the boosty library:

```
terminate called after throwing an instance of 'boost::filesystem::filesystem_error'
  what():  boost::filesystem::read_symlink: The request is not supported: "<a symlinked folder>"
```

Which doesn't seem to be defined. Does this possibly relate to  #1647 ?

kintel, 2016-06-07:
@Coolway99 Thanks for digging. That might indeed be related. Do you, by any chance, manage to get a stack trace showing where in the OpenSCAD code this crash originates?

Coolway99, 2016-06-07:
@kintel Yes and no. It doesn't explicitly create a stack trace when it crashes, but it seems to only crash after a critical point. When openscad is in a symlinked directory, it crashes like above. --help and --version don't crash it however and work like expected, but --info is an instant crash. The same crash happens when opening files that have a "include" to a symlinked file, including a symlinked file while scripting. I don't know how to debug it, as I haven't worked much with C++, but I hope what I've provided is of help.

PeterGrace, 2016-09-15:
I wanted to speak up and alert the thread that I'm also having the same problem on Windows 10.  OpenSCAD is installed locally on the disk, but my design files are junctioned to a different path on my filesystem.

If I move the files from the junctioned directory to a different directory, the problem does not manifest.

Let me know what I can do to help with troubleshooting.

```
C:\cnc\OpenSCAD\5WMount>OpenGL Program Validation results:
Validation successful.
terminate called after throwing an instance of 'boost::filesystem::filesystem_error'
  what():  boost::filesystem::read_symlink: The request is not supported: "C:/cnc"

```

kintel, 2016-09-16:
@PeterGrace Which version of OpenSCAD is this? Can you reproduce it using the latest development snapshot?

PeterGrace, 2016-09-16:
The version I was running was 2015-03.  I uninstalled that, installed the latest available from the snapshots folder at files.openscad.org (2016.08.18 git 73eda81) and it also exhibits the same behavior.

kintel, 2016-09-16:
Thx. A stacktrace would help. Not sure if that's possible with the pre-built binaries.
We probably need someone with a native Windows build env to take a look at this, as symlinks work fine on other platforms.

t-paul, 2016-09-16:
Could be https://svn.boost.org/trac/boost/ticket/10900 ?

PeterGrace, 2016-09-16:
@t-paul it sure sounds like it.  The error emitted is from Boost, so I think you've hit the target.

donbright, 2016-09-27:
golang seems to have figured a solution

https://github.com/golang/go/issues/5750

nemws1, 2018-06-20:
This error has been biting me in the behind lately (version 2015.03-2, which I know is old).  DL'ed and tried the latest build (2018.06.01) and although it doesn't crash, it still doesn't follow the symlink.

I use Cygwin with 'nativestrict' symlinks (both Windows and Cygwin work just fine with these symlinks).

Would be really nice to have this working!

[image]

volkswagenfeature, 2019-06-29:
I'd like to note that I'm having a extremely similar issue: 
My documents folder on the SSD I use as my C:\ drive is symlinked to a folder on a secondary spinning platter drive. 
As such, whenever SCAD tries to autosave, it throws a harmless error that breaks nothing. However, when I added 
`assert(brickMinor <= brickMajor, "Brick has bad dimensions!")`
I received the following error:
[image]
This does not force the application to close, but causes it to hang. I can still edit code, but nothing else works. Nothing renders, and everything except the code editor is unresponsive.

IzK666, 2021-02-19:
I'm using latest openscad (2021.01) and I've the same error as @volkswagenfeature 
In my case, I have a regular C drive (an SSD), but the path where I store the scad files (W:\Files) is a partition assigned to a folder instead of a drive.

Whenever I want to render the code and there is any error (missing ; for example), the app tries to backup (in C drive) and throws the error, which shouldn't be affected, as the path to the file is not used.
If there is no mistakes in the code, I can store the file or run the render without issues.

Dromantor, 2021-11-25:
Since the issue with the boost library ([as linked somewhere above](https://svn.boost.org/trac/boost/ticket/10900)) was opened on their old bug-tracking-system ...

Which is now deprecated and ignored since 2018 or so:
> https://svn.boost.org/trac10
>
> IMPORTANT This trac is now deprecated, please head over to ​https://github.com/boostorg and file issues and PR's there instead.

... I have the strong feeling this won't get fixed until its reported on their github.
I'll just open an issue there and point them to this issue and to their old issue tracker. Maybe this wil finally get fixed.

Dromantor, 2021-11-25:
There was a merged PR for boost in May 2020 that sounds somewhat relevant to this issue:
#boostorg/filesystem/pull/100
They even mentioned the issue in their old bug tracker and said this fixes it.
What version of boost is used for the builds one can download from your website? If its lower than 1.74.0, released in August 2020, this could be the cause of the issue.

t-paul, 2021-11-25:
There's currently still a build issue with lib3mf, but once that's fixed, windows builds should have boost 1.77. List of versions is available via Help->Library Info.

t-paul, 2021-11-27:
@Dromantor Windows builds are back up with boost 1.77.

Dromantor, 2021-11-28:
Just tested it and it launches fine now, and after testing with some example files* no severe errors occured. Great :)
* One issue remaining is that I cannot open any of the example files directly through the links in the GUI (i.e. the start-screen or file->examples) when going through a mounted folder (symlinks work fine), it just does nothing. Opening through file explorer or file->open works though.

t-paul, 2021-11-28:
Closed via boost 1.77 upgrade. Remaining issue moved into separate ticket #3988.

closed 2021-11-28 by t-paul

alastaira, 2023-05-29:
Not sure if there's been a recent regression or not, but I encountered this error message today using version 2021.01 on Windows 11. 
In my case, it happens whenever I try to <use> or <include> a library that is symlinked in the OpenSCAD libraries folder, which sounds like it should have been a resolved case from this thread?

t-paul, 2023-05-29:
Please try with the dev snapshot version. The Release pre-dates this fix.

alastaira, 2023-05-30:
Oh, sorry - my mistake - I mistakenly thought the latest stable version was from 2022 so assumed it had included this fix (despite the fact I can clearly see its version 2021.01!).
Can confirm that the problem seems to be resolved in the latest development snapshot (v2023.05.28).

Thankyou!

jverne, 2024-03-12:
I can report that I was seeing this message in 2021.01 on Windows 10 with any preview attempt:

`boost::filesystem::read_symlink: The request is not supported: "C:/Users\abcd"`

It would still render, but obviously complained about the error at the top of the enchanted 3D view. There was also a red log line in the console.

Reinstalling 2024.03.11 previews error free. No errors in the console.