openscad/openscad #1317: STL files not working

kosme, 2015-04-19:
Just moved from version 2014.03 to 2015.03 and suddenly all the STL files that are generated seem to contain invalid triangles that cause pycam to fail. If the same source file is compiled in 2014.03, the STL file that is generated works ok.

kintel, 2015-04-19:
Do you have a small example? Do you know if there are any limitation in pycam in terms of what types of triangles it can accept?

kosme, 2015-04-19:
Just read a file generated in 2015.03 and all the facet have the same vector 0 0 0, I guess the error is there. This is a link to the source code and the files generated in 2014.03 and 2015.03 https://www.dropbox.com/sh/m59aw39onbffnfx/AADwuA3BdnK9Dt6tlymJJGtAa?dl=0

PR #1319 (2015-04-21): Fixed bug reversing result of is_nan(), causing normals vectors to somet...
...ime be exported as 0 0 0. Fixes #1317
commit 167b8825 (2015-04-21): Fixed bug reversing result of is_nan(), causing normals vectors to sometime be exported as 0 0 0. Fixes #1317
closed 2015-04-21 by kintel