openscad/openscad #1320: Recursive Module FREEZES COMPUTER

Eibwen, 2015-04-21:
I copied the body of a module that called another module into that other module (there by a recursive module of itself, with no options changed) and it output an ERROR of recursive module detected... but then proceeded to use every bit of ram and virtual memory possible with nothing responding for 15 minutes until i finally held the power button to force shutdown.

Windows version 2015.03

Eibwen, 2015-04-21:
This should reproduce it:

	$fn=100;

	rotatorPeg();

	module originalModule()
	{
	    height=3;
	    pegR=1.5;
	    rivetedPart=1;
	    rivetedAmount=0.3;
	    
	    difference()
	    {
	        union()
	        {
	            //cylinder(h=height, r=outerR);
	            translate([0,0,-OVERLAP])
	                cylinder(h=height+2*OVERLAP, r=pegR);
	            translate([0,0,height-rivetedPart+OVERLAP])
	            cylinder(h=rivetedPart, r1=pegR, r2=pegR+rivetedAmount);
	            
	            copiedBody(height, pegR, rivetedPart, rivetedAmount);
	        }
	        
	        union()
	        {
	            
	        }
	    }
	}
	module copiedBody(height, pegR, rivetedPart, rivetedAmount)
	{
	    GAP=0.2;
	    difference()
	    {
	        union()
	        {
	            //cylinder(h=height, r=outerR);
	            translate([0,0,-OVERLAP])
	                cylinder(h=height+2*OVERLAP, r=pegR);
	            translate([0,0,height-rivetedPart+OVERLAP])
	            cylinder(h=rivetedPart, r1=pegR, r2=pegR+rivetedAmount);
	            
	            copiedBody(height, pegR, rivetedPart, rivetedAmount);
	        }
	        
	        union()
	        {
	            
	        }
	    }
	}

t-paul, 2015-04-21:
See #1276.

Please add a working test case to verify it's fixed with #1276.

kintel, 2015-04-21:
Closing as duplicate of #1276.

closed 2015-04-21 by kintel

MichaelAtOz, 2015-04-21:
Code above, when fixed, is trapped by the latest recursion trap on W7/64.