openscad/openscad #1321: Crash on Exporting STL

alafi, 2015-04-22:
The following model cause OpenSCAD (version 2015.03) to crash when exporting to STL file.

```
translate([10,10,0]){
  union(){
    BedTexture();
    translate([10,0,0]) BedTexture();
    translate([0,12,0]) BedTexture();
    translate([10,12,0]) BedTexture();
  }
}

module BedTexture(){
  difference(){
    union(){
      scale([2,2.5,1.3]) sphere(4,$fn=25);
      translate([5,6,3]) scale([1,1,0.5]) sphere(1,$fn=14);
      translate([-5,6,3]) scale([1,1,0.5]) sphere(1,$fn=14);
      translate([5,-6,3]) scale([1,1,0.5]) sphere(1,$fn=14);
      translate([-5,-6,3]) scale([1,1,0.5]) sphere(1,$fn=14);
    }
    difference(){
      cube([20,20,20], center=true);
      translate([0,0,5]) cube([10,12,10], center=true);
    }
  }
}
```

t-paul, 2015-04-22:
That's probably #1258. Could you verify if it's fixed with the snapshot version: http://files.openscad.org/snapshots/OpenSCAD-2015.04.13-x86-64.zip

alafi, 2015-04-22:
Yes, it is working fine in the provided snapshot.

Thank you very much!

t-paul, 2015-04-22:
Thanks for testing, an official bugfix release will happen soon.

closed 2015-04-22 by t-paul