openscad/openscad #1322: Can't change document path?

budney, 2015-04-23:
Hi!

OpenSCAD keeps creating the directory $HOME/Documents/OpenSCAD, with subdirectories "backups" and "libraries". I don't want it there, but there's no way to change it in the preferences.

I noodled around in the documentation, and tried editing ~/Library/Preferences/org.openscad.OpenSCAD.plist to add a key called NSDocumentDirectory with the path I want as a value, but it doesn't work as I'd hoped: it's still creating $HOME/Documents/OpenSCAD. I notice that whenever I open OpenSCAD, it overwrites org.openscad.OpenSCAD.plist and removes the key I added.

Is there a way to do this that I don't know about? If not, can this be fixed?

budney, 2015-04-23:
It's really, REALLY unmannerly for apps to stick files wherever they want, disregarding the wishes of the user.  :-(

kintel, 2015-05-04:
@budney Do you have a good suggestion to how we should allow specifying where to store (and search for) these files? Whenever we make such changes, we need to keep multi-platform operation into account, so solutions tend to be more complex than just doing it for one platform..

budney, 2015-05-04:
@kintel I'm not sure I follow the question: The path itself will be platform dependent, but handling file paths in a multi-platform way is something you already handle several places. For example, the file-open dialog, and the implementation of the import() function.

Your file-open dialog is multi-platform, and it remembers the location at which the last file was opened. I find that path stored in the .plist file on my Mac, but you could easily expose it in the preferences dialog.

Apparently the path for the libraries and backups is hard-wired in the code, and there are different paths for different platforms; this path should be exposed in the preferences. You can store the path as a string, let users set it in the preferences, and handle it using the same code you use to parse the last file-open location.

t-paul, 2015-12-17:
See proposal in #1524. I'd suggest to close this one and continue with the other issue.

budney, 2015-12-20:
Will it enable me to specify that my OpenSCAD files belong in
$HOME/Documents/O/OpenSCAD? If so, that's fine with me. If not, that
doesn't work for me--I need to put the files where I want them.

kintel, 2015-12-20:
@budney Yes, that proposal will allow changing the document location.

budney, 2015-12-20:
That sounds great. Thanks!

t-paul, 2019-11-22:
Closing, see #1524.

closed 2019-11-22 by t-paul