openscad/openscad #1324: read() function for 2D and 3D source into lists

clothbot, 2015-04-23:
As an alternative to purpose-specific module requests like bounds() (see issue #1088), please provide a read() function equivalent to the import() module (ideally using the same code-base) for reading import()-supported file formats into list structures. The resulting read() list should be polygon() and polyhedron() compatible with minimal user intervention.

Hypothetical usage:

// 2D:
my2Dobj=read("my2dmodel.dxf");
my2Dobj_points=my2Dobj[0];
my2Dobj_paths=my2Dobj[1];
polygon(points=my2Dobj_points,paths=my2Dobj_paths);

// 3D:
my3Dobj=read("my3dmodel.stl");
my3Dobj_points=my3Dobj[0];
my3Dobj_faces=my3Dobj[1];
polyhedron(points=my3Dobj_points,faces=my3Dobj_faces);

Motivation: Once the lists have been created, iterating over them for finding bounding boxes/cubes, proximity-based operations, fancy chain-hull operations on non-manifold source files, etc. becomes relatively trivial.

kintel, 2015-04-27:
Another, similar function would be to have a UI function for resolving imported files into literal lists.

clothbot, 2015-04-27:
Thinking out loud...

The resulting lists are often huge, significantly larger than the original source files. That's what I discovered via this Blender plugin I helped with: https://github.com/graphicsforge/io_mesh_openscad

...Being able to Export "SCAD List File" could be fun+easy to implement.  We'd probably want to pair it with a "Compress" option to zip-compress output and support include/use zip-compressed scad files.  Generate an uncompressed 'mySTL.scad' file to handle/reference the parts contained in the 'mySTL.zcad' compressed object.

As long as there's a way to bypass the manifold-ness requirement and provide command-line options to "parse 2D/3D input file to compressed SCAD output+header file", this could be quite handy.

atartanian, 2015-04-29:
the read() concept sounds really great to me. It would be very useful to have access to external files component (point, face) lists.

t-paul, 2016-02-20:
Naming wise, I'd like the solution suggested in #1585 - just use `import()`.

doug-moen, 2016-02-21:
Here's the proposed OpenSCAD2 syntax for this.

`import()` is a module that returns a shape. In OpenSCAD2, shapes are values. You can call a module from within an expression, you can assign a shape to a variable or pass a shape as a function argument. We introduce a set of built-in functions which query a shape and return various kinds of data. Here are some shape query functions:
* `is_shape(s)` -- true if s is a shape
* `is_2dshape(s)`, `is_3dshape(s)`
* `2d_points(s)` -- the points list for a 2D shape, which is a list of 2-vectors
* `2d_paths(s)` -- the paths list for a 2D shape
* `3d_points(s)`
* `3d_faces(s)`
* ... and more, as needed

For example,
```
// 2D:
my2Dobj = import("my2dmodel.dxf");
my2Dobj_points=2d_points(my2Dobj);
my2Dobj_paths=2d_paths(my2Dobj);
polygon(points=my2Dobj_points,paths=my2Dobj_paths);
```

ameuret, 2023-10-18:
> We introduce a set of built-in functions which query a shape and return various kinds of data.

That would open up so many possibilities !! 😍 Has this proposal ever moved forward?

closed 2026-11-03 by github-actions[bot]

jordanbrown0, 2026-11-03:
For the original request, I would think of extending the (currently-experimental) import("foo.json") function to support additional file types.  Some file types, like CSV and TXT, would be relatively easy.  Others, like DXF, SVG, and 3MF, would require considerable design care to ensure that all possible files are imported in a useful way.