openscad/openscad #1325: Tests not working

indazoo, 2015-04-24:
Hi,
I wanted to upgrade my threads library from for loops to list comprehensions and waited for the official release. Now, with the new version I started my code...BANG!!! Crash (horrible, if you like drama). Same with latest dev snapshot available on http://www.openscad.org/downloads.html 
Sigh...
Well, I saw some other people suffering in the issues and saw they executed the test suite. I tried to run it but it failed most of the time stating, that it did not found
 C:\Program Files (x86)\ImageMagick-6.9.1-Q16\modules\coders\IM_MOD_RL_PNG_.dll
Then, after googling I added "C:\Program Files (x86)\ImageMagick-6.9.1-Q16" to my User path and to my system path.
Both did not worked. I had to execute
>   SET PATH=%PATH%;C:\Program Files (x86)\ImageMagick-6.9.1-Q16
explicitly/directly in the cmd window opened by "OpenSCAD_Test_Console.py". Then the tests passed (gloriously, for even more drama).

The test on http://www.openscad.org/downloads.html is a bit old, and created some errors because of depreceated OpenScad syntax for which I think have already been done some bugfixing on this repo . I'm not sure where to add that path thing so I could not check if it is already fixed.So, sorry for double post.
I'll check back for a new version in a few months. Until then 2014.03 works fine for me.

kintel, 2015-04-24:
Did you test the 2015.03-1 update, which was released yesterday?
Also, do you have a (small) example which demonstrates this issue?

indazoo, 2015-04-24:
I downloaded from
http://www.openscad.org/downloads.html

I'm not sure, because yesterday is a wide time in this www world, but just to be sure, I downloaded it (2015.03-1 , x86, Windows, zip)  right now...<insert explosion sound>
Sorry
It fails right after "CSG Products generation"

Your request for a small sample...uh....my little son just dropped in,father time....a few hours perhaps..I'll look to put something together out of my threads library.
Until then, many thanks for your response

***************************************************
OpenSCAD Version: 2015.03-1
System information: Windows 7 SP1 (v6.1.1.0) 4 CPUs WOW64 7.68 GB RAM
Compiler: GCC "4.9.2" 32bit
Boost version: 1_57
Eigen version: 3.2.2
CGAL version, kernels: 4.5, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.0
Qt version: 5.4.1
QScintilla version: 2.8.4
MingW build: MingW32
GLib version: 2.42.1
Application Path: C:/Temp/OpenScad/openscad-2015.03-1
Documents Path: C:\Users\User\Documents
Resource Path: C:/Temp/OpenScad/openscad-2015.03-1
User Library Path: C:/Users/User/Documents/OpenSCAD/libraries
User Config Path: C:\Users\User\AppData\Local/OpenSCAD
Backup Path: C:/Users/User/Documents/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
C:/Users/User/Documents/OpenSCAD/libraries
C:/Temp/OpenScad/openscad-2015.03-1\libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
GLEW version: 1.12.0
OpenGL Version: 2.1.0 - Build 8.15.10.2476
GL Renderer: Intel(R) HD Graphics
GL Vendor: Intel
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Using QGLWidget

GL Extensions:
GL_3DFX_texture_compression_FXT1
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_copy_buffer
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_map_buffer_range
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_sampler_objects
GL_ARB_seamless_cube_map
GL_ARB_shader_objects
GL_ARB_shading_language_100
GL_ARB_shadow
GL_ARB_sync
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_non_power_of_two
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_transpose_matrix
GL_ARB_uniform_buffer_object
GL_ARB_vertex_array_bgra
GL_ARB_vertex_array_object
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_window_pos
GL_ATI_separate_stencil
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_clip_volume_hint
GL_EXT_compiled_vertex_array
GL_EXT_draw_buffers2
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_object
GL_EXT_gpu_program_parameters
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_packed_pixels
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_texture3D
GL_EXT_texture_array
GL_EXT_texture_compression_s3tc
GL_EXT_texture_edge_clamp
GL_EXT_texture_env_add
GL_EXT_texture_env_combine
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_lod_bias
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_shared_exponent
GL_EXT_texture_snorm
GL_EXT_texture_swizzle
GL_EXT_transform_feedback
GL_IBM_texture_mirrored_repeat
GL_INTEL_performance_queries
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_WIN_swap_hint

kintel, 2015-04-24:
One thing that might help: Your OpenGL drivers are almost 4 years old, coming from a vendor known for unstable drivers. A typical failure mode for drivers is to crash the running process. Try updating them - that has helped a lot of Intel customers so far.

indazoo, 2015-04-25:
Yeah, I checked that. Unfortunately I have a Intel Processor with GMA HD. No Updates above OpenGL 2.1. The Intel update tool says "you have a laptop with vendor specific drivers and I refuse to detect what drivers you need". Nice, not? 
After some earlier "adventures" with graphic card drivers of another vendor I don't think the grass is greener somewhere else.
Further more, the old OpenScad version runs OK. No problems there.
I have no idea what "super advanced" features you use to display 3D blocks. For me it looks like 1995 tech would suffice for OpenScad display needs. OK I know, things in Software always look easy first...the software trap....;-) Perhaps the nice new GUI has its price?
Seems like I have to buy a new computer to use OpenScad...well....I like OpenSource, but this is a bit too much. Seems like the effort I put into the threads library seems not to "pay" out. I wanted to give something back to OpenSource community with the library. But this is really hard. Especially after this other adventure I lately had with printrun. You have to install about 10-15 additional tools onto python and then on top of that a compiler and compile the whole thing. A few days work to find a way to run it. Incredible...
I'll probably drop the parametric idea and try something like DesignSpark or stay with the old version.
If OpenScad were written in C# I would help but I have no desire to battle with ancient C/C++ pointers. The battle you seem to loose...Why so many coders hurt themselves so much is really a mystery for me. Everything the evil force (MS) has thought of cannot be used. WE WANT SUFFER!!! WE HAVE A RIGHT TO SUFFER!!! 
Enough rants. I'll check it out in a few months.
PS. Simplifying my code is not possible, since it is a loop which connects polygons.

indazoo, 2015-04-25:
$fn=25;
test_threads();

module test_threads ($fa=5, $fs=0.1)
{
    // M8
    metric_thread(8, pitch=1.5, length=5);
    translate ([0, 15, 0])
        metric_thread(8, pitch=1.5, length=5, right_handed=false);
    // multiple start:
    translate ([0, -15, 0])
    metric_thread(8, pitch=3, length=5, internal=false, n_starts=3);

    translate ([10, 0, 0])
    square_thread(8, pitch=1.5, length=5);

    translate ([20, 0, 0])
    acme_thread(8, pitch=1.5, length=5);

    translate ([30, 0, 0])
    buttress_thread(8, pitch=1.5, length=5);

    translate ([40, 0, 0])
    english_thread(1/4, pitch=20, length=1/4);

}

module test_channel_threads()
{
	// channel thread
	translate ([10, 0, 0])
         test_channel_thread(8);
	
	translate ([-10, 0, 0])
         test_channel_thread2();
}

module test_slot_tabs()
{
	// tabs & slots
	translate ([0, 0, +5])
		test_tabs(ref_dia = 10);
	color("LemonChiffon")
	translate ([0, 0, -5])
		test_slots(ref_dia = 10);
}

module test_metric_thread ($fa=5, $fs=0.1)
{
	metric_thread( diameter = 20,
		pitch = 4, 
		length = 3, 
		internal=false, 
		n_starts=3, 
		right_handed=true,
		clearance = 0.1, 
		backlash=0.4,
		printify_top = false
	);
}
module test_metric_left()
{
	metric_thread(8, 
				pitch=1.5, 
				internal=false, 
				length=3, 
				right_handed=false);
}

module test_hollow_thread ($fa=5, $fs=0.1)
[... 2576 lines cut ...]
						translate([0,0,f_tabHeight])
							rotate([0,90,0])
								cylinder(r=lock, h=outer_radius, $fn=12);
					}
				}//end union

				// cuts right tab flank
				rotate(f_tabWidth_angle/2)
					translate([0,outer_radius,f_tabHeight/2])
						cube([2*(outer_radius+1), 2*outer_radius, f_tabHeight+0.001],center=true);
					
				// cuts left tab flank 
				rotate(-f_tabWidth_angle/2)
					translate([0,-outer_radius,f_tabHeight/2])
						cube([2*(outer_radius+1), 2*outer_radius, f_tabHeight+0.001],center=true);

				// subtract inner leftover per slot
				// make inner radius of tab smaller so it gets longer and
				// can cross the gap between tab and slot cylinder
				translate([0,0,-f_tabHeight])
				cylinder(r=inner_radius+join_distance(tabs_outward), h=f_tabHeight+2*f_tabHeight);

			}		
	}
}

t-paul, 2015-04-25:
@indazoo, why are you giving up just after the first suggestion that did not work? From the info output I see you are running the 32bit OpenSCAD version on 64bit Windows. Maybe try the 64bit one as the OpenGL driver might be different (small chance, but still worth a try, I guess).
An additional option is to try the Mesa OpenGL driver which is just a software renderer. This would not use any hardware acceleration, but OpenSCAD currently does not benefit from that much anyway (short description in the [manual](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Building_on_Microsoft_Windows#OpenGL_.28Optional.29), if you need some further info about how to install this driver, we can add some more details there...)

Just waiting a month will not fix the problem as it's related to the specific system/hardware which we don't have. So without any additional effort from your side, it's essentially impossible to help.

kintel, 2015-04-25:
Managed to reproduce. See #1326 for a fix.

indazoo, 2015-04-25:
@kintel
Wow, already fixed...!!! You have the pointers under control it seems...Congratulations!
Impressed...

@t-paul
I didn't gave up after first try. If that were the case, I would never had posted an issue here. Of course I tried 32/64 bit. To be honest, the issue I opened here was about the testing thing.I tried to find what failed to help this poor kintel guy who keeps the shop together as it seems.

It was more a culmination of anger after:
- hours fideling with netfabb showing errors in the perfectly valid OpenScad generated threads by my lib
- shortly followed by yelling bad words at my monitor over GSC use in OpenScad (I hate the necessity to create overlapping objects. really! ) I want objects which I can join in an object oriented manner !!! (stomping with my foot like my little son). 
- building printrun needed days until it ran correctly
- Slicer also was no help. This string parsing python monster I had to change/recode too and use another useless language invented by a bored university professor...
Damn! I just wanted to create a product!
Why must one do everything by oneself??? Sigh...Lions everywhere...no paradise ;-)
At least my Graphics card has now the latest driver. It didn't helped anyway. But cost me another 2 hours to find it (GMA, Intel HD Graphics, first generation processors, Arrandale, Ironlake) all these words I had to understand to get it. Man, if not all my friends would try to frantically install linux on their MACs I would buy one. There, the world is not purple too it seems.

t-paul, 2015-04-25:
I've uploaded new [snapshot builds](http://www.openscad.org/downloads.html#snapshots) for Windows that include the #1326 fix.

indazoo, 2015-04-26:
You two, Bitcoin account numbers please...

For me, all hope is lost anyway. 
Now, Atlassians SourceTree (Git Tool) does not recognize changes after line 1700. So i cannot commit them....
https://answers.atlassian.com/questions/15281655/sourcetree-ignores-changes-on-file-with-many-changes
My karma seems to be degraded. But I only stole one candy as a child!!! :-)
well....time to learn farming or something.

kintel, 2015-04-26:
The official OpenSCAD bitcoin donation address is listed on http://www.openscad.org/

closed 2015-04-27 by kintel

indazoo, 2015-04-28:
Bitcoins sent. Thanks you two.
For me the case is closed, I assume some day, the tests (initial issue) will be updated anyway.

kintel, 2015-04-28:
Thanks. Getting the test suite to run for Windows users is a larger task - pending a developer on the team who uses Windows on a daily basis :)