openscad/openscad #1328: Option color-scheme preference for scale marker

korpx, 2015-04-27:
Add preference for the color of the scale markers in color-schemes json-files

kintel, 2015-05-04:
FYI: Currently, the axis color is calculated from the background color using [ColorMap:: getContrastColor()](https://github.com/openscad/openscad/blob/master/src/colormap.cc#L238).

ArchimedesPi, 2015-05-12:
Just checking, but I'm assuming that the axis rendering you're talking about is in GLView.cc, and the colour is set around [here](https://github.com/openscad/openscad/blob/2001f1302e79ead7ac65a6915c812136a5c90391/src/GLView.cc#L161)?

t-paul, 2015-05-12:
Yes, that's where the color is calculated (and later used for the axis/scales and the letters from the small axis).

kintel, 2015-05-15:
Fixed by #1338

closed 2015-05-15 by kintel

korpx, 2015-10-19:
axes-color worked fine in 2015.09.03 but has no effect in 2015.10.17
https://github.com/openscad/openscad/blob/master/color-schemes/render/tomorrow-night.json

kintel, 2015-10-19:
It's because the latest compiled snapshot is a release candidate of 2015.03-2, which didn't get this feature backported. Will be fixed once we get 2015.03-2 out the door.