openscad/openscad #1330: nonplanar faces wrong displayed, F6 fails with assert violation

indazoo, 2015-04-30:
Hi,
Sorry, it's me again. Another Karma attack...

OpenScad 2015.04.25 git 2b33922

Code:

use <list-comprehension-demos/sweep.scad>					
TwoDShape = [[3.1881, 0], [4, 0.46875], [4, 0.65625], [3.1881, 1.125], [3.1881, 1.5], [4, 1.96875], [4, 2.15625], [3.1881, 2.625], [3.1881, 4.5], [0, 4.5], [0, 0]];
//polygon(TwoDShape);
transform = [[[1, 0, 0, 0], [0, -2.22045e-016, -1, 0], [0, 1, -2.22045e-016, -1.5], [0, 0, 0, 1]], [[0.866025, 1.11022e-016, 0.5, 0], [0.5, -1.92296e-016, -0.866025, 0], [0, 1, -2.22045e-016, -1.375], [0, 0, 0, 1]]];
sweep (TwoDShape, transform);

==> creates a nice 3 D object, no errors shown

But !!!!!
intersection() {
sweep (TwoDShape, transform);
translate([0,0,1])
cube(size = [8.8, 8.8, 1.4], center = true);
}

==> creates an error
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
PolySet has nonplanar faces. Attempting alternate construction
ERROR: Alternate construction failed. CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: pe_prev->is_border() || !internal::Plane_constructor::get_plane(pe_prev->facet(),pe_prev->facet()->plane()).is_degenerate() File: /data/OpenSCAD/libraries-mingw64-master/mxe-w64/usr/x86_64-w64-mingw32.static/include/CGAL/Nef_3/polyhedron_3_to_nef_3.h Line: 251 
Geometries in cache: 8
Geometry cache size in bytes: 9960
CGAL Polyhedrons in cache: 4
CGAL cache size in bytes: 0
Total rendering time: 0 hours, 0 minutes, 0 seconds
Rendering finished.

My two cents:
The polyhedron created by sweep() has some polygons with many corner points. These may be not planar. Or maybe the small numbers in the transform..
Here is the CSG Tree Dump

group() {
	intersection() {
		group() {
			polyhedron(points = [[3.1881, 0, -1.5], [4, -1.04084e-016, -1.03125], [4, -1.45717e-016, -0.84375], [3.1881, -2.49801e-016, -0.375], [3.1881, -3.33067e-016, 0], [4, -4.37151e-016, 0.46875], [4, -4.78785e-016, 0.65625], [3.1881, -5.82868e-016, 1.125], [3.1881, -9.99202e-016, 3], [0, -9.99202e-016, 3], [0, 0, -1.5], [2.76097, 1.59405, -1.375], [3.4641, 2, -0.90625], [3.4641, 2, -0.71875], [2.76097, 1.59405, -0.25], [2.76097, 1.59405, 0.125], [3.4641, 2, 0.59375], [3.4641, 2, 0.78125], [2.76097, 1.59405, 1.25], [2.76097, 1.59405, 3.125], [4.99599e-016, -8.65332e-016, 3.125], [0, 0, -1.375]], faces = [[0, 1, 12, 11], [1, 2, 13, 12], [2, 3, 14, 13], [3, 4, 15, 14], [4, 5, 16, 15], [5, 6, 17, 16], [6, 7, 18, 17], [7, 8, 19, 18], [8, 9, 20, 19], [9, 10, 21, 20], [10, 0, 11, 21], [10, 9, 8, 7, 6, 5, 4, 3, 2, 1, 0], [11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21]], convexity = 5);
		}
		multmatrix([[1, 0, 0, 0], [0, 1, 0, 0], [0, 0, 1, 1], [0, 0, 0, 1]]) {
			cube(size = [8.8, 8.8, 1.4], center = true);
		}
	}
}

indazoo, 2015-04-30:
The *%&/ç ui here deleted the use statement:
use   list-comprehension-demos/sweep.scad 
With the smaller/greater signs of course

kintel, 2015-04-30:
sweep() is work in progress, and it currently doesn't handle self-intersections very well. If you move your 2D object a little bit away from the X axis, it should work better.

indazoo, 2015-05-01:
Thanks for the feedback. It didn't worked better and to be honest, using overlaps is not really on top of my preferred things. Using Beta releases also not. So I dropped the sweep stuff and created the thread polyhedron with list comprehensions. Nice new feature of OpenScad! Rendering threads is now 10 times faster  in comparison to polyhedron concatenation. 
Case closed for me
Thanks!

kintel, 2015-05-04:
@indazoo If you could share your final thread model somewhere, that could be good inspiration for making a good example or demo.

indazoo, 2015-06-05:
The thread library is now in a reasonable state. Not sure if it fits as a sample, probably too big.
It is a single file you may or not include it into MCAD lib (as you like).
https://github.com/indazoo/MCAD/blob/master/fasteners/threads.scad

You can, in a more or less easy way, define your own tooth profile.
It is much faster than the stuff which is floating around.

Teaser:
 *   > Metric threads
 *   > Square threads
 *   > ACME threads
 *   > Buttress threads
 *   > Channel threads
 *   > Rope threads (for rope pulleys)
 *   > NPT, BSP  (for tapered pipes)
 *   > Simple twist and lock connectors
 *   
 *   > All can have a bore in the center
 *   > All can have multiple starts
 *   > All support internal(nut) and external(screw)
 *
 *   > Very fast rendering, no "normalization tree" problems

Enjoy

indazoo, 2015-06-05:
Ah... I dropped all use of sweep...

kintel, 2024-03-25:
Status as of 2024-03-24: This now works when using Manifold

closed 2026-02-02 by github-actions[bot]