openscad/openscad #1331: Add support for import / export 3MF format

clothbot, 2015-05-01:
Creating this issue to gather info and interest around the new 3MF file format:

* [What is 3MF](http://www.3mf.io/what-is-3mf/)
* [3MF Specification](http://www.3mf.io/what-is-3mf/3mf-specification/)

Cautiously optimistic that the standard will be as open as claimed...

t-paul, 2015-05-01:
Yeah, https://xkcd.com/927/ :-). But then it's not really new, it's just the already existing Microsoft format. I guess it remains to be seen what "The source code will be on Github and cross-platform code is in development." means. At least the licensing requirements for the source code sound reasonable and Microsoft does not try to destroy evil open source anymore (not so sure about some other consortium members) ;-).

clothbot, 2015-05-01:
Straw(man) poll over on [the mailing list](http://forum.openscad.org/New-3MF-file-format-td12525.html) is leaning toward wait'n'see.  (to put it politely... ;-)

clothbot, 2015-05-02:
As per the discussion on the mailing list, closing this ticket for the time being.

Near-future development activity needs to focus on multi-material and colour/texture support in the already implemented, basic AMF format support.

closed 2015-05-02 by clothbot

t-paul, 2016-09-20:
This may finally happen... #1802 :-).