openscad/openscad #1332: Render Issue with OpenSCAD Simple Problem

JCrew7384, 2015-05-03:
Hi. I am relatively new to OpenSCAD and having some issues in a simple script I wrote. I have an STL of Big Ben I downloaded from Thingiverse. I repaired it in Netfabb, it found some issues. I am trying to just build a small buffer around Big Ben and create half an ice mold. When I hit F5, it works perfectly. When I hit F6, it just shows me the cube, no Big Ben... Any idea?

// BIG BEN OBJECT 

SIZE
x = 19.02;
y = 19.02;
z = 113.72;

// MOLD BUFFER
in_buffer = 0.5;
mm_buffer = in_buffer * 25.4;

// CUBE SIZE
cubex = x + mm_buffer * 2;
cubey = y + mm_buffer * 2;
cubez = z + mm_buffer;

difference() {
    
translate([-cubex/4,0,cubez/2]) cube([cubex/2,cubey,cubez], center=true);

scale(2,2,2) translate([0,0,-0.5]) import("/Users/jordanmarton/Desktop/Big_Ben (Repaired).stl");

}

JCrew7384, 2015-05-03:
Link to Big Ben File: https://www.dropbox.com/s/pidctny3faptl6f/Big_Ben%20%28repaired%29.stl?dl=0

JCrew7384, 2015-05-03:
OpenSCAD Version: 2015.03-1
System information: Mac OS X Version 10.10.3 (Build 14D136) iMac11,3 4 CPUs 16.00 GB RAM 
Compiler: Clang "6.0 (clang-600.0.57)" 64bit
Boost version: 1_57
Eigen version: 3.2.4
CGAL version, kernels: 4.5.2, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.0
Qt version: 5.4.1
QScintilla version: 2.8.4
MingW build: No
GLib version: 2.42.1
Application Path: /Applications/OpenSCAD.app/Contents/MacOS
Documents Path: /Users/jordanmarton/Documents
Resource Path: /Applications/OpenSCAD.app/Contents/Resources
User Library Path: /Users/jordanmarton/Documents/OpenSCAD/libraries
User Config Path: /Users/jordanmarton/Library/Application Support/OpenSCAD
Backup Path: /Users/jordanmarton/Documents/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
/Users/jordanmarton/Documents/OpenSCAD/libraries
/Applications/OpenSCAD.app/Contents/Resources/libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
GLEW version: 1.12.0
OpenGL Version: 2.1 ATI-1.32.20
GL Renderer: ATI Radeon HD 5750 OpenGL Engine
GL Vendor: ATI Technologies Inc.
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Using QGLWidget

GL Extensions:
GL_APPLE_aux_depth_stencil
GL_APPLE_client_storage
GL_APPLE_element_array
GL_APPLE_fence
GL_APPLE_float_pixels
GL_APPLE_flush_buffer_range
GL_APPLE_flush_render
GL_APPLE_object_purgeable
GL_APPLE_packed_pixels
GL_APPLE_pixel_buffer
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_specular_vector
GL_APPLE_texture_range
GL_APPLE_transform_hint
GL_APPLE_vertex_array_object
GL_APPLE_vertex_array_range
GL_APPLE_vertex_point_size
GL_APPLE_vertex_program_evaluators
GL_APPLE_ycbcr_422
GL_ARB_color_buffer_float
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_imaging
GL_ARB_instanced_arrays
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_provoking_vertex
GL_ARB_seamless_cube_map
GL_ARB_shader_objects
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shadow
GL_ARB_shadow_ambient
GL_ARB_sync
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_non_power_of_two
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_transpose_matrix
GL_ARB_vertex_array_bgra
GL_ARB_vertex_blend
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_window_pos
GL_ATI_blend_equation_separate
GL_ATI_blend_weighted_minmax
GL_ATI_separate_stencil
GL_ATI_texture_compression_3dc
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_bindable_uniform
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_clip_volume_hint
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_draw_buffers2
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_geometry_shader4
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_texture_array
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_s3tc
GL_EXT_texture_env_add
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_mirror_clamp
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array_bgra
GL_IBM_rasterpos_clip
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_light_max_exponent
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SGI_color_matrix

t-paul, 2015-05-03:
The STL still has some Zero Faces (triangle that have no area) which cause problems in OpenSCAD.

See https://github.com/openscad/openscad/wiki/Frequently-Asked-Questions#why-is-my-imported-stl-file-only-showing-up-with-f5-but-not-f6 for a description how to remove those with MeshLab.

JCrew7384, 2015-05-03:
That was exactly the issue... How did you know? OpenSCAD provided no errors. Did you just download and go through a series of tests yourself? Or just try to run that test in MeshLab and see that it worked? Any advice to prevent future problems, or a series of steps you always take when downloading and importing an STL?

Thanks!

t-paul, 2015-05-03:
Yes, I downloaded the file and checked with the "3D Printing Toolbox" which highlights various issues. The Zero Faces problem is known from other reports, there are some ideas of how to fix some of those mesh issues on import, but this will take some time to implement.

JCrew7384, 2015-05-03:
What is the 3D Printing Toolbox? Is that the Blender Addin that I see in a
google search?

*Jordan Marton*

email *Jordan.Marton@gmail.com <Jordan.Marton@gmail.com> *| mob 617-905-0938

kintel, 2015-05-04:
This is probably a duplicate of #945

t-paul, 2015-05-04:
Yes, it's a plugin in blender which is shipped by default for some time, but it needs to be enabled in User Preferences.

t-paul, 2015-06-07:
Confirmed to be a duplicate of #945, so closing this one...

closed 2015-06-07 by t-paul