openscad/openscad #1333: center={other true values} not recognised

MichaelAtOz, 2015-05-04:
Re [forum](http://forum.openscad.org/allowing-center-parameter-to-be-a-vector-of-booleans-td12563.html#a12568), center= with values that elsewhere evaluate to true, e.g. <>0, "non null string", [{non empty set}], (apart from 'true') are not evaluated as 'true'. I don't know if this is intentional.

Note: If this is fixed, then there is a possible conflict with the proposal for center=[bool,bool,bool] as that would, if following the normal evaluation rules, be 'true'.

kintel, 2015-05-04:
We currently only consider values of type BOOL to be able to have a truthy value. Easy to change, but we should probably think before doing so..

MichaelAtOz, 2015-05-04:
Wiki says:
"Boolean Values

Boolean values are truth values. There are two Boolean values, named 'true' and 'false'. A Boolean value is passed as the **'center' argument** to the 'cube' and 'cylinder' primitives, as the argument to 'if', and as the arguments to the logical operators '!' (not), '&&' (and), and '||' (or).

**In all of these contexts**, you can actually pass any type of value. Most values, when used in a Boolean context, are equivalent to 'true', but there are a few that are equivalent to 'false'. The values that count as false are:

    false
    0 and -0
    ""
    []
    undef

Note that nan (Not A Number) counts as true."

MichaelAtOz, 2015-05-04:
```
b1=0;
b2=1;
b3=(b1==false); // this
b4=(b2==true);  // and this
b5=(!b1==true);
b6=(!b2==false);

ba=(b1?true:false);
bb=(b2?true:false);

echo(b1,b2,b3,b4,b5,b6,ba,bb);
b=[b1,b2,b3,b4,ba,bb];
for (i=[0:len(b)-1])
	translate([i*10,0,0])
   cylinder(d=10, h=10, center=b[i]);

translate([0,20,0])
	cylinder(d=10,h=10, center=!!0);
translate([0,-20,0])
	cylinder(d=10,h=10, center=!!1);
```
Gives
```
ECHO: 0, 1, false, false, true, true, false, true
```

Not in terms of the wiki, but just logical evaluation, I would have thought that the b3 & b4 evaluations (`b3=(b0==false)`) should be `true`. Anyway...

doug-moen, 2015-05-04:
I don't particularly like what the Wiki says about Boolean values. I think it is a bad idea to select an arbitrary representative from each value type and call that value 'false'. The arbitrariness and unpredictability bother me.
 * Why is NaN considered true, when the IEEE float standard says that NaN==NaN should be considered false?
 * What about range values (oops, we missed that type, maybe an empty range should be false?)
 * What about function values (which are forthcoming)?
 * What about shape values (which are forthcoming)? Should cube(0) be false?

The ideal design is to print an error message to the console whenever a value of the wrong type is used in an expression. For example, if a non-Boolean is used in a logical expression, then ideally, we should print at least a warning message.

I know that backward compatibility is a concern. However, the closer we can get to this ideal, then the better we are able to support extensions like cube(sz, center=[true,true,false]) without breaking backward compatibility. Extensions like this extend the domain of a function by admitting new argument values that were previously illegal or unsupported. If there was an error message in the illegal/unsupported case, then extending the domain changes the behaviour of a program that previously generated error messages. If there are no error messages, and there is an expectation that "any type of value can legally be used in a boolean context", say, then the extension becomes riskier, because people may write code depending on the old behaviour.

MichaelAtOz, 2015-05-07:
Strong typing in OpenSCAD would be a big negative step in my opinion. (not feeling humble today)

Its ease of use and learning for beginners is a major benefit. It already does some implicit  typecasting/coercion, I just think it should be consistent for Booleans. If you add '==' to the list of logical operators supported, and fix the center= to as described, it is consistent.

For function/shape values: undef=false, else=true.
p.s. How do you get an empty range?

doug-moen, 2015-05-07:
I'm not sure what you mean by "strong typing". I normally consider that to
mean "static type checking", which is not what I am advocating.

I agree that static type checking creates difficulties for beginners,
because you have to learn how to write your code in a way that makes the
static type system happy. Lots of natural seeming idioms are rejected by
the type checker. For example, if OpenSCAD were statically typed, then it
would be impossible to extend the center= parameter of cube() to accept an
array of booleans as an argument, because "Boolean" and "array of Boolean"
would be different types, and the static type system would force us to
choose a single unique type for center=.

OpenSCAD is a dynamically typed language, and it needs to stay that way,
precisely because of "ease of use" and "learning for beginners".

My basic position is that OpenSCAD should produce error messages when you
try to execute nonsense code, instead of failing silently and producing
unpredictable results. As a beginner, I had a problem where I kept typing
things like
   cube(1,2,3)
and then had to debug my program and find the error when my results were
not as expected. I should have typed cube([1,2,3]), but OpenSCAD won't tell
you when you make this kind of mistake: it ignores superfluous arguments,
won't tell you if you forgot to supply required arguments, and if arguments
have the wrong type, it just does something arbitrary and unpredictable
instead of telling you where the error is.

This behaviour that I'm complaining about does not promote ease of use, nor
does it help learning for beginners. I find it very frustrating.

cube(4,5,6) is equivalent to cube(4)
cube() is equivalent to cube(1)
cube("foo") is equivalent to cube(1)

Should we document all this behaviour in the wiki? Should we define a type
coercion rule so that "foo" is officially equivalent to 1, and then change
the equality operator so that "foo"==1 is true?

===
If we do change the equality operator the way you suggest, then the
problems I am describing will get much worse.

You want to change the equality operator so that

0==false is true
""==false is true
[]==false is true
undef==false is true

If all of these values are ==false, then do they also equal one another?
For example, what is 0==""? Should that be true? What about 0==undef?
Should that also be true?

If 0==false, then perhaps 2+false == 2.
If ""==false, then perhaps len(false) == 0.

Javascript went down a similar design path, where you extend every
operation to "make sense of" every type, and then formalize the resulting
ad-hoc equivalences as type coercion rules. This is generally regarded as a
major design flaw. Douglas Crockford has a nice discussion of this in his
book "Javascript: The Good Parts". There are two appendixes describing the
"Bad" and the "Ugly" parts of Javascript, where these type coercions
feature prominently.

With the == operator performing type coercions, it becomes fairly useless
for some of the things that OpenSCAD programmers need to do. The same
problem occurs in Javascript, but Javascript "fixes" that by providing a
=== operator that does the same thing that == does right now in OpenSCAD:
compare two values without doing type coercion. Crockford recommends that
you should never use the == operator, because it causes too many problems:
use === instead.

We could introduce the same design flaw into OpenSCAD, but I don't
recommend it.

If we formalize the bugs in the logic operators as standard type coercions,
then it will indeed be true that
  0==undef
  ""==undef
  []==undef
and that means you can no longer use "x==undef" to test if an expression
has failed to produce a meaningful result. Right now, arithmetic operations
fail and return undef if you pass a non-numeric argument, and you can test
for this failure using x==undef. But if 2+false returns 2 (instead of
undef) and 0==undef is true, then none of that works any more.

drogge, 2015-05-25:
I started to look into "fixing" the original "problem" that this issue addresses and decided to check out the comments on the issue and I'm glad I did. What I'm looking for in using other data types to represent bools is to have a more terse representation of true and false. I'd much rather be able to use center=[0,0,1] for the new centering option for cube than to have to write center=[false,false,true].

Although it may be that openscad has take type conversion to bool a bit far with allowing strings and lists to represent booleans I think that using numerical values should be find. Granted that things like nan can get a bit weird.

The argument that if we allow arbitrary data types to be converted to boolean would break the comparison operators is a bit of a red herring. The  == operator doesn't convert its operands automatically  to boolean. It just returns a boolean result. For example if we think that the == operator only compares booleans then if (0 == "') would be true. In the 2015.03 openscad it returns false. Granted I'm of the opinion that it should return a error but, as pointed out above, openscad doesn't like to complain about things :-) If openscad had explicit type conversions like bool(undef) or bool("") then one could do things like if (bool(0) == bool("")) but I'm not sure what the use of that would be.

I'd like to continue with ultimately adding the list option to the center operator but not until a decision is made on this issue. If them's that are in the power want to decide one way or the other then I'll see what I can do.

MichaelAtOz, 2015-05-26:
To address @doug-moen 's comment.

> (from further above) Why is NaN considered true, when the IEEE float standard says that NaN==NaN should be considered false?

Elsewhere in the wiki NaN is not equal to anything, even itself.

> My basic position is that OpenSCAD should produce error messages when you try to execute nonsense code, instead of failing silently and producing unpredictable results.

+1. But not directly on topic. (IMHO)

> If all of these values are ==false, then do they also equal one another? For example, what is 0==""? Should that be true? What about 0==undef? Should that also be true? 
>
> If 0==false, then perhaps 2+false == 2. 
> If ""==false, then perhaps len(false) == 0.

`0==""` true, `0==undef` (see below after AHHH). Using constants there is illogical, just like `true==false` - why would anyone code that. When you use variables it makes sense.
I am not proposing that boolean be coerced everywhere, just in the `==` operator. 
The last two examples would be 
`false` (2+false = undef, undef == 2 = false), and 
`true` (len(false) = undef, undef == 0 = true). But that is illogical so results are illogical.

> If we formalize the bugs in the logic operators as standard type coercions, then it will indeed be true that 
0==undef 
""==undef 
[]==undef
and that means you can no longer use "x==undef" to test if an expression has failed to produce a meaningful result. Right now, arithmetic operations fail and return undef if you pass a non-numeric argument, and you can test for this failure using x==undef. But if 2+false returns 2 (instead of undef) and 0==undef is true, then none of that works any more.

As above, not proposing coercion in expressions, just the result.
Again you are using constants, if you replace those with variables it will work.

**AHHH**, *I see.*

OK, I propose in addition to implementing boolean coercion for `==`, and (within '==') the constant `undef` never gets coerced. It will be the small exception proving the rule. (v's the large exception where `==` behaves differently to `? :` and `if`)
So (as I presumed) coercion only happens when types mismatch.  Any variable/expression is either undef (or NaN?), or has a value of another type. So any `<something>==undef` where `<something>` is undef will return true. `<something> else` will coerce to a boolean and as `undef` is not coerced, the `<boolean>==undef` result will be false (as it is clearly defined).

Thus for the above, 0==undef is (and would continue to be) false, as 0 would coerce to false and false==undef will be false, as will be the others above.
Hence `x==undef', where x is undef, will be true.

2+false would still be undef (as such should really be a logical expression as now occurs, 2 || false = true, 0 && true = false) [off topic]

If however you did:
```
t=1;
x=2+t;
```
you get what you wrote....

That still leaves the issue of center=[b,b,b]. Theoretically [b,b,b] would be true, but I would suggest that treating a 3-vector as a x/y/z would have precendent. i.e. center= 3-vector | Boolean | coerced-Boolean.

doug-moen, 2015-05-26:
drogge wrote: "What I'm looking for in using other data types to represent bools is to have a more terse representation of true and false. I'd much rather be able to use center=[0,0,1] for the new centering option for cube than to have to write center=[false,false,true].

Although it may be that openscad has take type conversion to bool a bit far with allowing strings and lists to represent booleans I think that using numerical values should be find. Granted that things like nan can get a bit weird."

There's nothing wrong in wanting terser code. If you do this:
  T = true;
  F = false;
then you can use center=[F,F,T].

I will try to present an objective case as possible for why we shouldn't have such complicated rules for "what counts as false" in a boolean expression.

It's been said on many occasions that OpenSCAD is a solid modelling tool, not a programming language. The design should serve the needs of 3D designers, and not assume programming experience. It shouldn't have unnecessary complexity just because that's what somebody is used to after learning some popular programming language.

I argue that having a long and arbitrary list of "values that count as false" adds unnecessary complexity, and is just adds to the list of arbitrary things you must memorize in order to use the language. I'm arguing that simplicity is the overriding design factor here.

MichaelAtOz, 2015-05-26:
What about adding T/F as 'constants' like PI. And maybe X=0,Y=1,Z=2 too...

drogge, 2015-05-26:
I hadn't considered just assigning to simple variables. I did think about functions for implementing T() and F(). Not for very long though. I wonder if there would be any backwards compatibility issues with using T and F? Kind of tromping on the user namespace.

Maybe in openscad2 we could just have true/false/zero/non-zero as bools?

MichaelAtOz, 2015-05-26:
Here is the code doing it (I think), from src\value.cc
```
bool Value::toBool() const
{
switch (this->type()) {
case BOOL:
return boost::get<bool>(this->value);
break;
case NUMBER:
return boost::get<double>(this->value)!= 0;
break;
case STRING:
return boost::get<std::string>(this->value).size() > 0;
break;
case VECTOR:
return boost::get<VectorType >(this->value).size() > 0;
break;
case RANGE:
return true;
break;
default:
return false;
break;
}
}
```

doug-moen, 2016-01-25:
Runsun just posted that cube(10, center=1) doesn't center the cube. That's also true for square.

This relates to a different beef that I have with cube and square. As I beginner, I had trouble learning to type cube([5,10,20]), and I typed cube(5,10,20) instead. Ditto with square: instead of square([5,10]) I typed square(5,10). My beef is that there is no error message in these cases. What I would like is for cube and square to abort with an error message if you pass the wrong number or the wrong types of arguments.

In order for this to work the way I want, it's necessary to abort if the 2nd argument, which is the 'center' argument, is not a boolean. Otherwise, if we interpret integers as booleans, then square(5,10) is treated the same as square(5, center=true), which is just confusing.

closed 2026-02-02 by github-actions[bot]