openscad/openscad #1334: Multibyte character problem with Tibetan

tamarintech, 2015-05-04:
Using the text() module I am unable to reproduce text as expected.
Using the common Jomolhari font I am trying to reproduce the popular "om mani padme hum" text.  In the screenshot below you can see the text in LibreOffice displaying correctly.  In the OpenScad editor the text also appears correctly.  The rendered text, however, does not.  The first and last characters are not displayed correctly.

openscad 2015.05.03, Fedora 21.

[image]

kintel, 2015-05-04:
Two ways to reproduce:
```
text("ཨོཾ", font="Jomolhari");
translate([0,14]) text("\u0f68\u0f7c\u0f7e", font="Jomolhari");
```

kintel, 2015-05-04:
This blog post shows very similar issues: http://tom.to/blog/archives/76

t-paul, 2015-05-04:
Setting the `script` parameter `"Tibt"` will select the correct shaping (naming comes from harfbuzz which uses https://en.wikipedia.org/wiki/ISO_15924 script tags). I guess should look at other programs how they select this automatically (maybe based on the unicode area?).
```
text("ཨོཾ", font="Jomolhari", script = "Tibt");
```

closed 2015-05-04 by tamarintech