openscad/openscad #1336: Is there something im doing wrong in the following code

MrRebel, 2015-05-08:
I am getting a non 2 manifold error when trying to export the following code as an .stl . Why is this?
	
	module bearing(){
		rotate([0,90,0]){
			cylinder(24,7.5,7.5,true,$fn = 100  );
			cylinder(50,5,5,true);
			difference(){
				cylinder(4,9,9,true);
				cylinder(4,8,8,true);
			}
		}
	}
	
	module fullcar(){
		difference(){
			cube([26,34,14],true);
			cylinder(16,6,6,true,$fn=100);
			translate([0,0,4.85]){
				cylinder(3.7,8,8,true,$fn = 100);
			}
			translate([0,0,-4.5]){
				cylinder(3,8,8,true,$fn = 100);
			}
			translate([0,-17,-5]){
				bearing();
			}
			translate([0,17,-5]){
				bearing();
			}
			translate([10,8,0]){
				cylinder(16,1.5,1.5,true,$fn = 100);
			}
			translate([10,-8,0]){
				cylinder(16,1.5,1.5,true,$fn = 100);
			}
			translate([10,0,0]){
				cylinder(16,1.5,1.5,true,$fn = 100);
			}
			translate([-10,0,0]){
				cylinder(16,1.5,1.5,true,$fn = 100);
			}
			translate([12,0,3]){
				cube([7,7,2.5],true);
			}
			translate([-12,0,3]){
				cube([7,7,2.5],true);
			}
		}
	}
	fullcar();

MichaelAtOz, 2015-05-08:
It is common for the flat sides of cylinders to have a problem of a common edge with a cube, particularly near a tangent.

One fix is:
```
module bearing(){
    rotate([0,90,0]){
        cylinder(24,7.5,7.5,true,$fn = 100  );
        cylinder(50,5,5,true,$fn=100); // fn 
        difference(){
	    translate([-0.1,0,0])  // keep edges separate (not the prob. in this case, but good practice)
                cylinder(4,9,9,true,$fn=100); // fn
            cylinder(4.2,8,8,true,$fn=100); // fn & height to keep edge separate.
        }
    }
}
```

MrRebel, 2015-05-08:
hmmm doesn't seem to fix it could any thing else be an issue?

MichaelAtOz, 2015-05-08:
Could you paste your bearing code here, it works for me. Also what OpenSCAD version & platform used?

MichaelAtOz, 2015-05-08:
Hang on, now it is not working....

MichaelAtOz, 2015-05-08:
It still is the flat sides of the cylinder and the cube, this misalignes them by a smidgen.
```
        translate([0,-17.01,-5]){
            bearing();
        }
        translate([0,17.01,-5]){
            bearing();
        }
```
Export worked this time.

MrRebel, 2015-05-08:
Perfect i finally gave up trying to fix it myself after 3 days of working on it. Thanks so much.

closed 2015-05-08 by MrRebel

MichaelAtOz, 2015-05-08:
I found it, elimentary

The clue:
[image]

The flat surface of the hole is co-incident with the flat face of the bearing cut-out.