openscad/openscad #1340: CGAL assert error (only under Valgrind)

ArchimedesPi, 2015-05-13:
So I decided to run Valgrind on OpenSCAD for "fun".
My copy of OpenSCAD, which I grabbed from Git and compiled, works fine normally.
However, when I run it with `valgrind --suppressions=valgrind.supp ./openscad`, I get the following error:
````
==22846== Memcheck, a memory error detector
==22846== Copyright (C) 2002-2013, and GNU GPL'd, by Julian Seward et al.
==22846== Using Valgrind-3.10.0.SVN and LibVEX; rerun with -h for copyright info
==22846== Command: ./openscad
==22846== 
terminate called after throwing an instance of 'CGAL::Assertion_exception'
  what():  CGAL ERROR: assertion violation!
Expr: -CGAL_IA_MUL(-1.1, 10.1) != CGAL_IA_MUL(1.1, 10.1)
File: /usr/include/CGAL/Interval_nt.h
Line: 209
Explanation: Wrong rounding: did you forget the  -frounding-math  option if you use GCC (or  -fp-model strict  for Intel)?
==22846== 
==22846== HEAP SUMMARY:
==22846==     in use at exit: 56,176 bytes in 286 blocks
==22846==   total heap usage: 352 allocs, 66 frees, 65,895 bytes allocated
==22846== 
==22846== LEAK SUMMARY:
==22846==    definitely lost: 0 bytes in 0 blocks
==22846==    indirectly lost: 0 bytes in 0 blocks
==22846==      possibly lost: 5,727 bytes in 96 blocks
==22846==    still reachable: 50,449 bytes in 190 blocks
==22846==         suppressed: 0 bytes in 0 blocks
==22846== Rerun with --leak-check=full to see details of leaked memory
==22846== 
==22846== For counts of detected and suppressed errors, rerun with: -v
==22846== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
Aborted
````
I'm using the default Ubuntu libcgal-dev and libcgal10. Any ideas?

t-paul, 2015-05-13:
Valgrind does not support some of the rounding modes CGAL is using.

It's possible to disable the check using:
```
qmake DEFINES+=CGAL_DISABLE_ROUNDING_MATH_CHECK=ON
```

closed 2015-05-13 by t-paul

ArchimedesPi, 2015-05-13:
Thanks @t-paul, I was incredibly confused.

t-paul, 2015-05-13:
Oh, it took me some digging to get to the easy solution - mainly due to working on #709 which is still not fully solved.