openscad/openscad #1342: Syntax Highlighting Does Not Work on Linux Mint 17.1

sunjay, 2015-05-13:
My editor does not syntax highlight.

[image]

You'll notice that "Color Syntax Highlighting" is "Off". That dropdown has no other options. According to @kintel this is usually because the color schemes did not download for some reason.

My `dpkg -L` output is below. You'll notice that the editor color schemes are there.

	$ dpkg -L openscad
	/.
	/usr
	/usr/share
	/usr/share/man
	/usr/share/man/man1
	/usr/share/man/man1/openscad.1.gz
	/usr/share/appdata
	/usr/share/appdata/openscad.appdata.xml
	/usr/share/bug
	/usr/share/bug/openscad
	/usr/share/openscad
	/usr/share/openscad/locale
	/usr/share/openscad/locale/es
	/usr/share/openscad/locale/es/LC_MESSAGES
	/usr/share/openscad/locale/es/LC_MESSAGES/openscad.mo
	/usr/share/openscad/locale/cs
	/usr/share/openscad/locale/cs/LC_MESSAGES
	/usr/share/openscad/locale/cs/LC_MESSAGES/openscad.mo
	/usr/share/openscad/locale/fr
	/usr/share/openscad/locale/fr/LC_MESSAGES
	/usr/share/openscad/locale/fr/LC_MESSAGES/openscad.mo
	/usr/share/openscad/locale/de
	/usr/share/openscad/locale/de/LC_MESSAGES
	/usr/share/openscad/locale/de/LC_MESSAGES/openscad.mo
	/usr/share/openscad/locale/ru
	/usr/share/openscad/locale/ru/LC_MESSAGES
	/usr/share/openscad/locale/ru/LC_MESSAGES/openscad.mo
	/usr/share/openscad/examples
	/usr/share/openscad/examples/Old
	/usr/share/openscad/examples/Old/example009.scad
	/usr/share/openscad/examples/Old/example020.scad
	/usr/share/openscad/examples/Old/example016.scad
	/usr/share/openscad/examples/Old/example007.dxf
	/usr/share/openscad/examples/Old/example008.dxf
	/usr/share/openscad/examples/Old/example019.scad
	/usr/share/openscad/examples/Old/example011.scad
	/usr/share/openscad/examples/Old/example002.scad
	/usr/share/openscad/examples/Old/example006.scad
	/usr/share/openscad/examples/Old/example021.scad
	/usr/share/openscad/examples/Old/example014.scad
	/usr/share/openscad/examples/Old/example008.scad
	/usr/share/openscad/examples/Old/example009.dxf
	/usr/share/openscad/examples/Old/example004.scad
	/usr/share/openscad/examples/Old/example003.scad
	/usr/share/openscad/examples/Old/example022.scad
	/usr/share/openscad/examples/Old/example012.scad
	/usr/share/openscad/examples/Old/example010.dat
	/usr/share/openscad/examples/Old/example013.scad
	/usr/share/openscad/examples/Old/example017.scad
	/usr/share/openscad/examples/Old/example015.scad
	/usr/share/openscad/examples/Old/example007.scad
	/usr/share/openscad/examples/Old/example013.dxf
	/usr/share/openscad/examples/Old/example010.scad
	/usr/share/openscad/examples/Old/example023.scad
	/usr/share/openscad/examples/Old/example012.stl
	/usr/share/openscad/examples/Old/example005.scad
	/usr/share/openscad/examples/Old/example016.stl
	/usr/share/openscad/examples/Old/example024.scad
	/usr/share/openscad/examples/Old/example018.scad
	/usr/share/openscad/examples/Old/example001.scad
	/usr/share/openscad/examples/Basics
	/usr/share/openscad/examples/Basics/text_on_cube.scad
	/usr/share/openscad/examples/Basics/logo.scad
	/usr/share/openscad/examples/Basics/CSG-modules.scad
	/usr/share/openscad/examples/Basics/LetterBlock.scad
	/usr/share/openscad/examples/Basics/projection.scad
	/usr/share/openscad/examples/Basics/projection.stl
	/usr/share/openscad/examples/Basics/rotate_extrude.scad
	/usr/share/openscad/examples/Basics/logo_and_text.scad
	/usr/share/openscad/examples/Basics/linear_extrude.scad
	/usr/share/openscad/examples/Basics/CSG.scad
	/usr/share/openscad/examples/Advanced
	/usr/share/openscad/examples/Advanced/GEB.scad
	/usr/share/openscad/examples/Advanced/surface_image.scad
	/usr/share/openscad/examples/Advanced/surface_image.png
	/usr/share/openscad/examples/Advanced/animation.scad
	/usr/share/openscad/examples/Advanced/children_indexed.scad
	/usr/share/openscad/examples/Advanced/module_recursion.scad
	/usr/share/openscad/examples/Advanced/offset.scad
	/usr/share/openscad/examples/Advanced/children.scad
	/usr/share/openscad/examples/Functions
	/usr/share/openscad/examples/Functions/functions.scad
	/usr/share/openscad/examples/Functions/recursion.scad
	/usr/share/openscad/examples/Functions/polygon_areas.scad
	/usr/share/openscad/examples/Functions/list_comprehensions.scad
	/usr/share/openscad/examples/COPYING-CC0.txt
	/usr/share/openscad/examples/examples.json
	/usr/share/openscad/color-schemes
	/usr/share/openscad/color-schemes/render
	/usr/share/openscad/color-schemes/render/tomorrow.json
	/usr/share/openscad/color-schemes/render/nature.json
	/usr/share/openscad/color-schemes/render/monotone.json
	/usr/share/openscad/color-schemes/render/deepocean.json
	/usr/share/openscad/color-schemes/render/sunset.json
	/usr/share/openscad/color-schemes/render/solarized.json
	/usr/share/openscad/color-schemes/render/beforedawn.json
	/usr/share/openscad/color-schemes/render/tomorrow-night.json
	/usr/share/openscad/color-schemes/render/metallic.json
	/usr/share/openscad/color-schemes/render/starnight.json
	/usr/share/openscad/color-schemes/editor
	/usr/share/openscad/color-schemes/editor/solarized-dark.json
	/usr/share/openscad/color-schemes/editor/tomorrow.json
	/usr/share/openscad/color-schemes/editor/visualstudio.json
	/usr/share/openscad/color-schemes/editor/light-background.json
	/usr/share/openscad/color-schemes/editor/tomorrow-night.json
	/usr/share/openscad/color-schemes/editor/dark-background.json
	/usr/share/openscad/color-schemes/editor/monokai.json
	/usr/share/openscad/color-schemes/editor/solarized-light.json
	/usr/share/openscad/color-schemes/readme.txt
	/usr/share/applications
	/usr/share/applications/openscad.desktop
	/usr/share/pixmaps
	/usr/share/pixmaps/openscad.png
	/usr/share/doc
	/usr/share/doc/openscad
	/usr/share/doc/openscad/TODO.txt.gz
	/usr/share/doc/openscad/RELEASE_NOTES.gz
	/usr/share/doc/openscad/README.md.gz
	/usr/share/doc/openscad/changelog.Debian.gz
	/usr/share/doc/openscad/copyright
	/usr/share/mime
	/usr/share/mime/packages
	/usr/share/mime/packages/openscad.xml
	/usr/bin
	/usr/bin/openscad
	/usr/share/doc/openscad/examples

System information:
[image]

t-paul, 2015-05-13:
Hmm, looks like bad choice for resource folder detection in https://github.com/openscad/openscad/blob/master/src/PlatformUtils.cc#L59

Can you create the folder `/usr/share/openscad/libraries` (can be empty) and see if that fixes the issue?

sunjay, 2015-05-13:
I created the folder and the color schemes immediately became available. Thanks!

I did have to go and explicitly change from "Off" to one of the color schemes though since my preferences had already saved from before.

That's a nice workaround, what needs to be done to fix it?

t-paul, 2015-05-13:
We just need to check for a folder which is always distributed along with the application, probably `color-schemes`, at the code line linked above.

closed 2015-05-15 by kintel

sunjay, 2015-05-15:
Yay!

commit 3b4ec917 (2015-05-15): Fix resource path search; the 'libraries' folder isn't always installed. Fixes #1342
commit 031e016d (2015-05-15): Fix resource path search; the 'libraries' folder isn't always installed. Fixes #1342