openscad/openscad #1344: Mac: unexpectedly reverting progress back a few minutes

earboxer, 2015-05-14:
Sometimes, when I save or hit the enter key (like in find), it reverts my progress back a few minutes.

ArchimedesPi, 2015-05-14:
There are major problems with the progressbar - see #810.

t-paul, 2015-05-15:
Please add some more details.

- Which version of OpenSCAD are you using (maybe paste the output of "Help->Library Info")?
- Which Operating System (and which version)?
- What does "revert my progress" mean?
- Does it happen with all kind of files or only when using `include`/`use`?

Maybe check if "Design->Automatic Reload and Preview" is enabled which triggers on save. Some older version of OpenSCAD had some issues related to that.

t-paul, 2015-06-07:
@earboxer can you please give some more info about the issue?

earboxer, 2015-06-16:
OpenSCAD Version: 2015.03-1
System information: Mac OS X Version 10.10.3 (Build 14D136) MacBookPro12,1 2 CPUs 16.00 GB RAM 
Compiler: Clang "6.0 (clang-600.0.57)" 64bit
Boost version: 1_57
Eigen version: 3.2.4
CGAL version, kernels: 4.5.2, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.0
Qt version: 5.4.1
QScintilla version: 2.8.4
MingW build: No
GLib version: 2.42.1
Application Path: /Applications/OpenSCAD.app/Contents/MacOS
Documents Path: /Users/earboxer/Documents
Resource Path: /Applications/OpenSCAD.app/Contents/Resources
User Library Path: /Users/earboxer/Documents/OpenSCAD/libraries
User Config Path: /Users/earboxer/Library/Application Support/OpenSCAD
Backup Path: /Users/earboxer/Documents/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
/Users/earboxer/Documents/OpenSCAD/libraries
/Applications/OpenSCAD.app/Contents/Resources/libraries

Revert my progress: the code changes to what it was several minutes ago.
Happens with all kinds of files, even very simple ones without includes.
Design-> Automatic Reload and Preview is on. I haven't tried with it off.

closed 2019-01-25 by t-paul