openscad/openscad #1347: working with STL - preview and render very different, F6 -> assertion violation 1102

kaefert, 2015-05-16:
When modifying STL files, some STL files look like I can modify them as long as I use preview (F5) but when I use render (F6) I see the original STL without my modifications again.

When combining multiple STL files everything works as expected as long as I use only preview (F5) - but rendering (F6) fails with those:

```
ERROR: CGAL error in CGALUtils::applyBinaryOperator difference: CGAL ERROR: assertion violation! Expr: itl != it->second.end() File: /usr/include/CGAL/Nef_3/SNC_external_structure.h Line: 1102 
ERROR: CGAL error in CGALUtils::applyBinaryOperator union: CGAL ERROR: assertion violation! Expr: itl != it->second.end() File: /usr/include/CGAL/Nef_3/SNC_external_structure.h Line: 1102
```

kintel, 2015-05-16:
This is a common problem with STL-files which are not well-formed, see:
https://github.com/openscad/openscad/wiki/Frequently%20Asked%20Questions#why-is-my-imported-stl-file-only-showing-up-with-f5-but-not-f6

kaefert, 2015-05-16:
brilliant! thanks for the hint, that workaround
```
MeshLab -> Filters -> Cleaning and Repairing -> Remove T-Vertices by Edge-Flip
```
worked nicely for me!

closed 2015-05-16 by kaefert