openscad/openscad #1349: PPA version doesn't work on xubuntu 14.04

mattcaron, 2015-05-21:
Installed via:
sudo add-apt-repository ppa:openscad/releases
sudo apt-get install openscad

Running with:
openscad

does nothing. It just hangs.

Running with:
strace openscad

Shows:
recvmsg(7, 0x7fff150c6840, 0)           = -1 EAGAIN (Resource temporarily unavailable)
poll([{fd=6, events=POLLIN}, {fd=7, events=POLLIN}, {fd=10, events=POLLIN}, {fd=11, events=POLLIN}], 4, 20803) = 1 ([{fd=6, revents=POLLIN}])

openscad --info says:

(matt@hiro) ~$ openscad --info
OpenSCAD Version: 2015.03-1
System information: Linux 3.13.0-53-generic #88-Ubuntu SMP Wed May 13 18:10:29 UTC 2015 x86_64 Ubuntu 14.04.2 LTS 4 CPUs 15.38 GB RAM
Compiler: GCC "4.8.2" 64bit
Boost version: 1_54
Eigen version: 3.2.0
CGAL version, kernels: 4.2, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.3.2
Qt version: 5.2.1
QScintilla version: 2.8.1
MingW build: No
GLib version: 2.40.2
Application Path: /usr/bin
Documents Path: /home/matt/.local/share
Resource Path: /usr/share/openscad
User Library Path: /home/matt/.local/share/OpenSCAD/libraries
User Config Path: /home/matt/.config/OpenSCAD
Backup Path: /home/matt/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  /home/matt/.local/share/OpenSCAD/libraries
  /usr/share/openscad/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:

GLEW version: 1.10.0
OpenGL Version: 3.0 Mesa 10.1.3
GL Renderer: Mesa DRI Intel(R) Ivybridge Mobile 
GL Vendor: Intel Open Source Technology Center
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
GL context creator: GLX
PNG generator: lodepng
GLX version: 1.4
OS info: Linux 3.13.0-53-generic #88-Ubuntu SMP Wed May 13 18:10:29 UTC 2015
Machine: x86_64

Version of OpenSCAD in the default 14.04 repos works fine.

I'm going to speculate that there's some fancy system daemon that you expect to be there on standard Ubuntu but is missing on Xubuntu.

kintel, 2015-05-21:
That's odd. Could you try the nightly build? That one is actually build specifically for XUbuntu - would be nice to know if it behaves differently: http://www.openscad.org/downloads.html#snapshots

mattcaron, 2015-05-21:
openscad-nightly works just fine... and looks so pretty...

kintel, 2015-05-21:
@hyperair Any idea what could cause this?

hyperair, 2015-05-22:
Not sure. Could you link me to your nightly build scripts? I could check what the differences are..

t-paul, 2015-05-22:
Main difference is probably Qt4 vs. Qt5.

Build scripts: https://build.opensuse.org/package/show/home:t-paul/OpenSCAD

hyperair, 2015-05-22:
Ugh. What do you think, @kintel? Should I switch back to Qt4? There's also that bug with smooth-but-delayed navigation like HD videos being played on old hardware without frame-dropping enabled.

t-paul, 2015-05-22:
Hmm, maybe I can try setting up a Qt5 build too to see if it's some kind of strange build env issue.

t-paul, 2015-07-04:
I've installed xubuntu-14.04 64bit in a VM (with all the current updates) and the OpenSCAD release works fine. Maybe it's related to the graphics driver.

Can you try running OpenSCAD with:
```
LIBGL_ALWAYS_SOFTWARE=1 openscad
```

This should change the OpenGL related `--info` output to something like 
```
GL Renderer: Gallium 0.4 on llvmpipe (LLVM 3.5, 256 bits)
GL Vendor: VMware, Inc.
```

mattcaron, 2015-07-05:
I need to test this theory, but I encountered a bug in IceWM (which I use because Xfwm4 windowshading is broken), where it wouldn't render windows from QT5. I'll retest with Xfwm4 and see if it is the same bug.

mattcaron, 2015-07-11:
Yeah, it's this bug in IceWM:

http://sourceforge.net/p/icewm/bugs/909/

If I don't use IceWM, it works. 

Closing.

closed 2015-07-11 by mattcaron