openscad/openscad #1352: xray view

tcurdt, 2015-05-27:
OpenSCAD is so amazing - but one thing that I am really missing is a xray/transparent preview

  https://youtu.be/0NnfNtErh20?t=18

That would make it soooo much easier to debug the models.

The wireframe is close - but not the same.

Also: Why is the wireframe only available for the rendered view - not the preview?

t-paul, 2015-05-27:
Preview has not enough information to display the mesh/wireframe as it's not calculated. The preview is based on clever rendering provided by the [opencsg library](http://www.opencsg.org/).

Transparent view for the rendered (F6) model should be possible, I'm not sure if there's a way for the preview.

The clipping planes view discussed in https://github.com/openscad/openscad/pull/1300 might catch at least some cases where looking at inside/hidden parts of a model is needed.

tcurdt, 2015-05-29:
An interactive clipping plane would be also very cool, indeed.

As for the preview - I frankly have not looked under the hood yet.
Maybe the opencsg library could be convinced to support transparency?

jordanbrown0, 2025-09-01:
You can use `color()` to make things transparent in preview.  You can use the `#` modifier to make them pink-and-transparent (and a bit magic).  You can use `%` to make things gray-and-transparent and magic in a different way.

tcurdt, 2026-08-31:
unstale