openscad/openscad #1353: Confirmation before rendering

kwendenarmo, 2015-05-28:
Sometimes you're designing, and instead of pressing F5, you pressing F6. Cancel the rendering is not immediately (in fact has never worked for me, I have finished saving the file, closing OpenSCAD and opening it again).
That is why, to avoid this, you could add a confirmation message that says something like, "This may take some time, are you sure?" when F6 is pressed.

MichaelAtOz, 2015-05-28:
Cancelling works for me, after a slight delay. On Intel Duo, Windows 7/64 w OpenSCAD many recent versions.
In theory the render is in another thread, so the GUI should be responsive.(as you say you can save etc)
What platform are you using, it could be a multi-thread issue??

kwendenarmo, 2015-05-29:
Linux Mint 17 3.13.0-24-generic #47-Ubuntu SMP x86_64 GNU/Linux on Intel(R) Core(TM) i7-4700MQ with OpenSCAD version 2015.03, compiled from sources.
However, although the option of canceling rendering works, the feeling of "doh" by pressing F6 by error, is simply avoidable.
You know what I mean, right?

MichaelAtOz, 2015-05-29:
Are you saying that the cancel button does work?

kwendenarmo, 2015-05-30:
Nop, I'm saying that even if canceling works, it would be nice a confirmation dialog before rendering.

Anecdotally, a few months ago, when he thought the cancel button works with a small delay, I pressed it before leaving home. Returning at home after five or six hours, OpenSCAD still rendering without having canceled. I had to kill the process.

tinkerology, 2015-05-30:
I often experience minutes of delay trying to cancel a complex render on
Windows 7 64bit, even with a recent snapshot build.

Scott
@tinkerology

nophead, 2015-05-30:
Yes I think it only cancels with the same granularity that the status bar updates in. The last operation can be a massive implicit union that takes most of the time and can't be cancelled once it starts I think.