openscad/openscad #1355: openscad_nogui requires -lX11

chrysn, 2015-05-29:
building the test suite failed on debian sid with:

    Linking CXX executable openscad_nogui
    /usr/bin/cmake -E cmake_link_script CMakeFiles/openscad_nogui.dir/link.txt --verbose=1
    /usr/bin/c++   -g -O2 -fPIE -fstack-protector-strong -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2  -fno-strict-aliasing -DEIGEN_DONT_ALIGN -O2 -g    -fPIE -pie -Wl,-z,relro -Wl,-z,now CMakeFiles/openscad_nogui.dir/home/chrysn/debian_packaging/openscad/openscad/src/openscad.cc.o  -o openscad_nogui -rdynamic libtests-offscreen.a libtests-cgal.a -lGLEW -lopencsg libtests-common.a libtests-core.a -lGLU -lGL -lglib-2.0 -lfontconfig -lfreetype -lharfbuzz -lboost_program_options -lboost_filesystem -lboost_regex -lCGAL -lboost_thread -lboost_system -lpthread -lgmp -lmpfr 
    /usr/bin/ld: libtests-offscreen.a(OffscreenContextGLX.cc.o): undefined reference to symbol 'XGetErrorText'
    //usr/lib/x86_64-linux-gnu/libX11.so.6: error adding symbols: DSO missing from command line
    collect2: error: ld returned 1 exit status

i managed to get it to build by adding `LINK_FLAGS "-lX11"` to the `set_target_properties(openscad_gui PROPERTIES...` line in tests/CMakeLists.txt as a workaround, but that will likely not work with other operating systems, and should be better set up where `OffscreenContextGLX.cc` is selected in the platform specific settings.

donbright, 2015-06-01:
i installed debian sid and it won't even get that far. 

```
[ 69%] Building CXX object CMakeFiles/tests-cgal.dir/home/don/openscad/src/cgalutils.cc.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.9/README.Bugs> for instructions.
CMakeFiles/tests-cgal.dir/build.make:284: recipe for target 'CMakeFiles/tests-cgal.dir/home/don/openscad/src/cgalutils.cc.o' failed
make[2]: *** [CMakeFiles/tests-cgal.dir/home/don/openscad/src/cgalutils.cc.o] Error 4
CMakeFiles/Makefile2:280: recipe for target 'CMakeFiles/tests-cgal.dir/all' failed
make[1]: *** [CMakeFiles/tests-cgal.dir/all] Error 2
Makefile:85: recipe for target 'all' failed
make: *** [all] Error 2
```

donbright, 2015-06-01:
so apparently it takes >2GB of RAM now on GCC 2.9.1 to build. 

i was able to reproduce your error using clang (which uses less RAM)

donbright, 2015-06-01:
found an interesting link.  apparently this might related to an update in Cmake's "FindOpenGL" code. 

https://bugs.webkit.org/show_bug.cgi?id=141796

https://bugzilla.redhat.com/show_bug.cgi?id=1194260#c3

http://www.cmake.org/pipermail/cmake/2015-March/060082.html

* The "FindOpenGL" module no longer explicitly searches for any
  dependency on X11 libraries with the "FindX11" module.  Such
  dependencies should not need to be explicit. Applications using X11
  APIs themselves should find and link to X11 libraries explicitly.|"

OK. Makes sense I guess. Lots of Non-X11 OpenGL coming to unix-land . . . working on a patch...

chrysn, 2015-06-01:
On Sun, May 31, 2015 at 07:47:30PM -0700, don bright wrote:
> found an interesting link.  apparently this might related to an update in Cmake's "FindOpenGL" code. 
> 
> https://bugs.webkit.org/show_bug.cgi?id=141796

the opengl parts might easily be the reason for why this is becoming an
issue now, but i wouldn't blame them -- if they don't mandate X11 any
more, why link against it.

rather, OffscreenContextGLX.cc explicitly uses XGetErrorText, and thus,
whatever adds it to the objects list should also add -lX11 to the
libraries list.

> so apparently it takes >2GB of RAM now on GCC 2.9.1 to build.

i'm hoping you meant 4.9.1.

the 2gb limit could easily hit the debian build servers as well (we
don't cross compile from amd64 to x86). without having dug too deeply
into it, this might be related to the test suite still being statically
linked.

some releases ago (with the advent of the command line export), we've
managed to get most of the test running from the regular openscad
binary. is that much of a (static) test suite still required?

t-paul, 2015-06-01:
I think it's basically `cgalutils.cc` which causes the huge memory impact, probably due to the big amount of template stuff coming in from CGAL.

I'm getting occasional crashes on [OBS](https://build.opensuse.org/package/show/home:t-paul/OpenSCAD) too where it gets stuck when compiling that file.

Statistics from Ubuntu 15.10 build, the numbers are quite steadily increasing with every new distribution (while compiling the same code base for the nightly builds).

*i586*
* Maximal used disk space: 1339 Mbyte 
* Maximal used memory: 1751 Mbyte  
* Total build: 1398 s

*x86_64*
* Maximal used disk space: 1485 Mbyte 
* Maximal used memory: 2545 Mbyte 
* Total build: 2048 s

kintel, 2015-06-01:
We could perhaps split cgalutils into separate files to limit the number of template expansions per compilation unit. Not sure how much it would help though.

donbright, 2015-06-02:
@Chrysn yes exactly.. my in-progress patch is doing a cmake findpackage on X11 for the offscreentest.cc for un*x machines. in the future, if we figure out non-X11 OpenGL on un*x, this could be modified hopefully without too much trouble.  im testing it out on some virtual machines that have 2 GB RAM thats why im assuming i hit the 2G limit crash (and some googling for gcc crashses for that 4.9.1 version. yes i did mean 4.9.1 not 2.9.1 lol thanks. )

@kintel  @t-paul i was able to build with gcc on a 2GB machine after splitting cgalutils.cc into three separate files. it is still hitting at least 1.5 GB gcc and a lot more for AS (assembler). AS is running paralell w GCC because cmake puts '-pipe' into the build line. (i even got the infamous Warning: end of file not at end of a line;" assembler crash)

https://forum.qt.io/topic/22672/g-crash-when-compiling-qt5/3

https://bugs.gentoo.org/show_bug.cgi?id=256766

i am working on branch named issue1355

hyperair, 2015-06-07:
Here's mine: http://anonscm.debian.org/cgit/users/hyperair/openscad.git/tree/debian/patches/Add-missing-X11-libraries-when-using-GLX.patch?h=debian

donbright, 2015-06-10:
thanks @hyperair

t-paul, 2015-07-04:
fixed by @donbright in 7279c1c019e014d12d8acfadd1a1f407c6da7f0c

closed 2015-07-04 by t-paul