openscad/openscad #1360: graphicsmagick's convert is not supported

chrysn, 2015-06-04:
the test suite depends on the `convert ... -alpha` option, which is supported in imagemagick's `convert`, but not in graphicsmagick's. that is generally documented (graphicsmagick forked from imagemagick at 5.5.2, while doc/testing.txt requires imagemagick 6.5.9.3), but gm's implementation is becoming widespread and several programs start depending on `convert

kintel, 2015-06-04:
How about ```-morphology``` Did graphicsmagick end up implementing that?

chrysn, 2015-06-04:
afaict not; it has been argued[1] that most of the things doable with -morphology can be done with -convolve as well with barely notable visual differences.

i've had a closer look at the -morphology use in the test suite, and found that the test suite *does* have the capability to detect whether -morphology (or `convert`) is available at all. it seems to be confused, though, by the presence of a `convert` binary. (that it is graphicsmagick seems to be detectable by ImageMagick_VERSION_STRING being empty). moreover, even if imagemagick is detected as absent, `DIFFPNG 1` is set too late for the `add_executable(diffpng diffpng.cpp ../src/lodepng.cpp)` line.

[1] http://sourceforge.net/p/graphicsmagick/discussion/250738/thread/da1ec46e/

donbright, 2015-06-05:
diffpng is extremely experimental (i wrote it to deal with weird systems where imagemagick was very broken). if you can do something with graphicsmagick it would probably be much easier and faster.

thehans, 2017-12-06:
@chrysn Since you say its missing `-alpha Off` support, does that mean GraphicsMagick always includes alpha in its convert?

I'm working on adding alpha support to png export, which would involve turning on alpha in the tests anyways.  Is there problems you would foresee with having alpha On?