openscad/openscad #1361: unit tests should be runnable out of tree

chrysn, 2015-06-04:
in order to find graphics card (or otherwise system dependent) regressions, it would be practical to have the test suite runnable out-of-tree.

most of the infrastructure for this is already available in the debian package, and i can provide a script to do that. (it is platform dependent so far because it expects things to be installed to a standard location, but that can still be tweaked).

the biggest show stopper in my opinion are the OPENSCAD_TESTING differences between openscad and openscad_nogui. (i'm opening a separate bug for that), and i'd gather the rest of the issues here.

donbright, 2015-06-10:
this may be also helpful to building a Windows(TM) unit test package. i am very interested to see this.

chrysn, 2015-06-10:
as of simplifying the debian package (basically, i found out that the automatisms that detect which build system is used and run things automatically can be used twice, to once detect and run the the qmake, and then to detect, run and test the cmake), i found that ctest is very well capable of running a test suite outside its own build directory; just the openscad scripts used don't support that.

my current approach in debian is to symlink everything required for the scripts to run into a testing directory (an approach that won't help much with windows), but on the long run, i'd prefer an [approach where none of the small scripts tinker with paths](https://github.com/openscad/openscad/pull/1363#issuecomment-109906409), and let ctest figure things out.

t-paul, 2015-08-18:
I just tried with:
```
cd /tmp/ && mkdir build && cd build/
cmake /data/OpenSCAD/openscad-master/tests
make -j8
ctest -j2
```
which seems to have worked fine except producing two warnings as it could not determine the git repo when creating the report.

Though, due to re-using the `openscad` binary produced by the `qmake` based main build process, that's not self-contained anymore.

t-paul, 2022-01-19:
Should be solved by the switch to `cmake`.

closed 2022-01-19 by t-paul