openscad/openscad #1362: remove OPENSCAD_TESTING build flag

chrysn, 2015-06-04:
the OPENSCAD_TESTING flag currently tweaks openscad's behavior to give reproducible results in unit tests. since 4d18db6 was merged, its use has been greatly reduced, but the remaining uses are:

* disabling building of the QT gui
* dropping the user library path
* normalizing ECHO output (removing timestamp=, normalizing floats)

as unit tests are nowadays run through python scripts, the normalization can happen there. dropping the user library path could be handled with a command line option (or people should just not have breaking things in their library when they run unit tests), and QT can stay (because then, we wouldn't need to build openscad_nogui for testing any more at all).

i can offer a patch that does all the normalization in test_cmdline_tool.py. then, we can drop the OPENSCAD_TESTING ifdefs in the code, and stop building openscad_nogui for the tests (and use the regular openscad instead).

this is a prerequisite for #1361.

chrysn, 2015-06-04:
a patch series is available on http://anonscm.debian.org/git/collab-maint/openscad.git in the chrysn/drop-openscadtesting branch -- can you work from that / review that or should i fork openscad on github and send a github internal pull request?

t-paul, 2015-06-04:
Only getting an empty page at that link....

t-paul, 2015-06-04:
Ahh, ok, that only works as git clone url.

chrysn, 2015-06-07:
as noted on irc, my patches erroneously removed the ENABLE_EXPERIMENTAL flag (which is still useful to build some binaries in tests/); moreover, those recently failed to build because the build system and the binaries rely on being built without QT (previously implicit with OPENSCAD_TESTING, now explicit with OPENSCAD_NOGUI). my branch has been updated to fix that.

@t-paul, there is also a web version available of the branch at http://anonscm.debian.org/cgit/collab-maint/openscad.git/log/?h=chrysn/drop-openscadtesting .

kintel, 2015-06-07:
@chrysn If you push your patches to github and d a pull request, it would simplify reviewing and applying this.

rewolff, 2015-07-09:
chrysn, fyi, I have had a need for the "-nogui" version. having it built/tested may help keep it in working order.

closed 2018-11-10 by t-paul