openscad/openscad #1364: Allow colours in SVG export

thinkl33t, 2015-06-09:
It would be very handy for Laser Cutter users if OpenSCAD's SVG export could export colours for different objects.  Differing fill colours are often used in laser cutting to separate objects into cut and mark profiles.

My thought was to use the existing colour definitions within OpenSCAD, assuming all that information isn't thrown away as part of the CGAL rendering process.  If no colour is specified for an object it could default to the grey that is currently being used.

    color("red") square([50,50]);

Would give:
[image]

bobc, 2015-06-14:
Colors currently do not survive the CGAL rendering process, so it needs enhancement / replacement of the CGAL backend.

lucasgelfond, 2015-07-01:
I understand how it would be a big fix but having color export option would be FANTASTIC.

mickvav, 2016-01-07:
I've made some coding on this. (https://github.com/mickvav/openscad/commit/493409033d4e55ab45ad8b37a3bf8bfc306a427c)

But it increases memory footprint and solves your problem in a different way - 
you can do, say

color("red") projection(cut=true) translate([0,0,-5])  cube([10,10,10]);

kaefert, 2017-04-13:
@mickvav Hello! I would also like to export a colored 2D object from openscad into svg format.
Could you explain to me what I need to do to take advantage of the work you referenced?
Has that work already been merged into the latest openscad release? Or is there some snapshot build somewhere that I can use? Or do I need to build your fork myself?
Thanks!

mickvav, 2017-04-13:
It wasn't pulled upstream, so yes - in general, you need to rebuild it.
It's not so difficult - just install the requirements and follow the
instructions, AFAIR. Those export-related things were not causing merge
conflicts, so you should be able to merge them with current upstream
relatively easily.

13 апр. 2017 г. 22:44 пользователь "Thomas" <notifications@github.com>
написал:

@mickvav <https://github.com/mickvav> Hello! I would also like to export a
colored 2D object from openscad into svg format.
Could you explain to me what I need to do to take advantage of the work you
referenced?
Has that work already been merged into the latest openscad release? Or is
there some snapshot build somewhere that I can use? Or do I need to build
your fork myself?
Thanks!

kaefert, 2017-04-14:
thanks @mickvav for the prompt reply! I will try that in the next days and report back on my results :)

JamesNewton, 2017-10-23:
This is a critical feature to allow the use of OpenSCAD with laser cutters. The Full Spectrum units distinguish between different settings by line color. E.g. the "red" line should be cut with 90% power and the "blue" line with 10% power, allowing both etching and cutting in one job. Actually, just being able to set the color of the line as a parameter of the projection would allow this. e.g. 
    projection(linecolor="red") modulename();

Other laser cutters use the line width... which would need to be a parameter of projection as there is no concept of line width in the 3D space.

I guess I should open a different feature suggestion issue for the idea of having parameters.

Metzen2k, 2020-02-14:
See this feature request for a similar idea (but with a different approach):
https://github.com/openscad/openscad/issues/3219

Metzen2k, 2020-02-19:
I think the first half of my comments in #3219 still apply (having an "exploded" SVG export option for laser cutting), even though #3219 was closed due to the 2nd half of the comments about color SVG's.

t-paul, 2020-02-19:
I don't know what "exploded" SVG means. This might be covered by the more general layer discussion #2383.

Metzen2k, 2020-02-19:
> I don't know what "exploded" SVG means. 

An SVG exported from OpenSCAD is one big path (or several big paths if you have more than one object). If you open the attached OpenSCAD SVG in Inkscape you can't highlight just the number and change their fill or stroke.

If you run this command using the attached PHP script:
php breakApartOpenScadSvg.php terminalPlate.svg > terminalPlateExploded.svg

Then you have an SVG where all of the paths are separate. When you open it up in Inkscape you can easily lasso the numbers and change the stroke color to red, for instance. Then it's easy in your laser cutter software (I'm using a GlowForge) to set the numbers to score (just darken on the surface) and the rest to cut.

I'm calling that an "exploded" SVG. I'm sure there's a better name for it. What I'm requesting / proposing is that OpenSCAD have a new "Exploded SVG" (or whatever) export format in OpenSCAD, which would make life much easier for laser cutter users, without changing things at all for existing OpenSCAD SVG users.

[breakApartOpenScadSvg.php.zip](https://github.com/openscad/openscad/files/4226928/breakApartOpenScadSvg.php.zip)
[terminalPlateExploded.svg.zip](https://github.com/openscad/openscad/files/4226931/terminalPlateExploded.svg.zip)
[terminalPlate.svg.zip](https://github.com/openscad/openscad/files/4226932/terminalPlate.svg.zip)

nophead, 2020-02-19:
> I'm using a GlowForge

I might have been but I got banned from the Glowforge forums for 1000 years
so I cancelled my order!

crozone, 2022-08-03:
I've written a small script that helps with the process of exporting SVGs/PDFs for laser cutting, it's designed for Trotec workflows but I assume it's similar for all laser cutters:

https://github.com/crozone/openscad-trotec-export

The basic strategy is to add `EXPORT_LAYER` variable to the part which controls which layer inside the part is enabled. The script then exports the part to SVG several times, setting `EXPORT_LAYER` to 1, 2, 3 etc. `EXPORT_LAYER = 0` is for preview and exports all layers.

The exported SVG is then parsed, the path extracted, and then modified to set the stroke color, stroke width, and fill color as required by the Trotec software. The SVG paths of all layers are combined back into a single SVG.

Optionally Inkscape is then called to convert the SVG into PDF since this is the preferred input format for Trotec.

In order to achieve the same results inside OpenSCAD natively, you cannot just set the `color()`. Full control over all of the path parameters, including stroke color, stroke width, and fill color is required.

burunduk3, 2023-06-18:
An alternative to the color control might be an option to control [SVG groups](https://developer.mozilla.org/en-US/docs/Web/SVG/Element/g), at least some laser cutting software is able to filter according them.

Though it's not clear to me how should it work, given that when laser cutting we think on toolpaths (edges), which openscad's main thing is to boolean geometries.

jordanbrown0, 2023-06-18:
With respect to lines versus shapes, see [OEP7: Mixed Dimension Geometry Support](https://github.com/openscad/openscad/wiki/OEP7%3A-Mixed-Dimension-Geometry-Support).

With respect to colors (or any other properties, like SVG groups) surviving render, see #1608 "color(), material() and part()".

crozone, 2023-06-19:
> With respect to lines versus shapes, see [OEP7: Mixed Dimension Geometry Support](https://github.com/openscad/openscad/wiki/OEP7%3A-Mixed-Dimension-Geometry-Support).

Does this cover OpenSCAD being able to export smooth curves vs a set of straight lines?

Currently OpenSCAD 2D is also sub-optimal for laser cutting because everything it exports is a polygon, it cannot export curves. I've found that sufficiently high `$fn` values will cause laser cutters to slow down and even resonate as they step through cutting a sufficiently detailed circle. So far the only fix is to use choppier circles or to try and post-process the SVG by interpolating the points back into a curve, which I've had limited success in doing.

jordanbrown0, 2023-06-19:
> Does this cover OpenSCAD being able to export smooth curves vs a set of straight lines?

Doubtful.

For 3D objects I expect that that will be Really Hard.  In my limited knowledge, none of the formats support curves.

For 2D objects it would still be Hard.  The underlying library used for 2D operations, Clipper, doesn't support curves.  It would be necessary to find or develop a library that can do boolean operations (union, intersection, difference) on curves.

But this is really orthogonal to the color feature that is the core of this issue.  Let's move line/curve discussion over into one of the issues that specifically discusses lines and curves - perhaps #99.

kintel, 2024-07-19:
3D color export was implemented in #5185. 2D color support is not yet supported by the geometry engines we use (Manifold only supports 3D geometry and Clipper doesn't support colors or face/edge attribute preservation).

AaronVerDow, 2025-07-30:
This pull request adds the ability to manually select output colors, but it still cannot automatically detect colors defined in the .scad file: https://github.com/openscad/openscad/pull/6052

gsohler, 2025-07-30:
Yes, I have been trying to work tofether for more than two years now.
My post was not meant to spam this thread but to show potentional goals and algorithms.
I do like the idea of "programming" shapes a lot,  I just dont agree on the speed, features and ideas become reality(e.g. this one dates back 10 years) whereas my dream is to have a versatile X-CAD with all features combined WITHIN my lifetime :)
And yes - I do accept the fact that others have very limited time to work on that, but PR's dont appear to speed up either ...
...being more silent in the future ...