openscad/openscad #1368: intersection not working

tcurdt, 2015-06-14:
I am trying to get the bevel gears working from http://www.thingiverse.com/thing:3575 but for some reason the intersection is not working.

In `bevel_gear` there is a   `difference` / `intersection` / `union` construct. Even after stripping it down to

    intersection () {
      union() {
        i = 0;
        // for (i = [1:number_of_teeth])
          rotate ([0,0,i*360/number_of_teeth])
            involute_bevel_gear_tooth (
              back_cone_radius = back_cone_radius,
              root_radius = root_radius,
              base_radius = base_radius,
              outer_radius = outer_radius,
              pitch_apex = pitch_apex,
              cone_distance = cone_distance,
              half_thick_angle = half_thick_angle,
              involute_facets = involute_facets);
      }

      translate ([-1.5*outside_pitch_radius,-1.5*outside_pitch_radius,0])
      cube ([3*outside_pitch_radius,
      3*outside_pitch_radius,
      bevel_gear_flat_height]);
    }

the intersection is not showing at all. Replacing the `intersection` with `union` shows that it actually should work:

[image]

tcurdt, 2015-06-14:
This is with version 2015.03-1 on OSX 10.10.3

t-paul, 2015-06-14:
Could you please post a compilable version with all the needed variables?

tcurdt, 2015-06-14:
Sorry for the long paste

    // Parametric Involute Bevel and Spur Gears by GregFrost
    // It is licensed under the Creative Commons - GNU GPL license.
    // © 2010 by GregFrost
    // http://www.thingiverse.com/thing:3575

    // Simple Test:
    //gear (circular_pitch=700,
    //	gear_thickness = 12,
    //	rim_thickness = 15,
    //	hub_thickness = 17,
    //	circles=8);

    //Complex Spur Gear Test:
    //test_gears ();

    // Meshing Double Helix:
    //meshing_double_helix ();

    // Demonstrate the backlash option for Spur gears.
    //test_backlash ();

    // Demonstrate how to make meshing bevel gears.
    bevel_gear_pair ();

    pi=3.1415926535897932384626433832795;

    //==================================================
    // Bevel Gears:
    // Two gears with the same cone distance, circular pitch (measured at the cone distance) 
    // and pressure angle will mesh.

    module bevel_gear_pair (
    	gear1_teeth = 41,
    	gear2_teeth = 7,
    	axis_angle = 45,
    	outside_circular_pitch=1000)
    {
    	outside_pitch_radius1 = gear1_teeth * outside_circular_pitch / 360;
    	outside_pitch_radius2 = gear2_teeth * outside_circular_pitch / 360;
    	pitch_apex1=outside_pitch_radius2 * sin (axis_angle) + 
    		(outside_pitch_radius2 * cos (axis_angle) + outside_pitch_radius1) / tan (axis_angle);
    	cone_distance = sqrt (pow (pitch_apex1, 2) + pow (outside_pitch_radius1, 2));
    	pitch_apex2 = sqrt (pow (cone_distance, 2) - pow (outside_pitch_radius2, 2));
    	echo ("cone_distance", cone_distance);
    	pitch_angle1 = asin (outside_pitch_radius1 / cone_distance);
    	pitch_angle2 = asin (outside_pitch_radius2 / cone_distance);
    	echo ("pitch_angle1, pitch_angle2", pitch_angle1, pitch_angle2);
    	echo ("pitch_angle1 + pitch_angle2", pitch_angle1 + pitch_angle2);

    	rotate([0,0,90])
    	translate ([0,0,pitch_apex1+20])
    	{
    		translate([0,0,-pitch_apex1])
    		bevel_gear (
    			number_of_teeth=gear1_teeth,
    			cone_distance=cone_distance,
    			pressure_angle=30,
    			outside_circular_pitch=outside_circular_pitch);
	
    		rotate([0,-(pitch_angle1+pitch_angle2),0])
    		translate([0,0,-pitch_apex2])
    		bevel_gear (
    			number_of_teeth=gear2_teeth,
    			cone_distance=cone_distance,
    			pressure_angle=30,
    			outside_circular_pitch=outside_circular_pitch);
    	}
    }
[... 573 lines cut ...]
    			rim_width = 5,
    			hub_thickness = 17,
    			hub_diameter=15,
    			bore_diameter=5,
    			backlash = backlash,
    			circles=8);
		
    		rotate ([0,0,360/teeth/4])
    		gear (
    			number_of_teeth = teeth,
    			circular_pitch=700,
    			gear_thickness = 12,
    			rim_thickness = 15,
    			rim_width = 5,
    			hub_thickness = 17,
    			hub_diameter=15,
    			bore_diameter=5,
    			backlash = backlash,
    			circles=8);
    	}

    	color([0,0,128,0.5])
    	translate([0,0,-5])
    	cylinder ($fn=20,r=backlash / 4,h=25);
    }

bobc, 2015-06-14:
If you select Thrown Together (F12), you will see that there are polygons with the wrong winding order. They are generated in involute_bevel_gear_tooth(), and reversing the order should help.
I.e. instead of 
                        faces=[[0,1,2],
            [0,2,3],
put
                        faces=[[2,1,0],
            [3,2,0],

tcurdt, 2015-06-15:
@bobc That was really a head-scratcher for me.
Didn't know one can see the order like that. Perfect! Thanks!

closed 2015-06-15 by tcurdt

nophead, 2015-06-15:
This comes up all the time. Is there a reason why it can't always show the
inside of the object in purple, rather than just the thrown together mode?

tcurdt, 2015-06-15:
+1

that or some other indicator would have helped a lot