openscad/openscad #1371: Syntax simplification

Steinhagelvoll, 2015-06-20:
Some commands can get really long. So it would be nice if one could omit the * sign, like it is done in math:

2*r = 2r

kintel, 2015-06-20:
This is likely to open a large wormhole of confusion and ambiguous syntax - I'd rather not go there.
..unless someone authors a comprehensible document covering all cases and manages to convince us it's actually a good idea :)

t-paul, 2015-06-20:
I think it's not really possible with the current syntax as it's conflicting with the exponential notation.
```
echo(2e3); ==> ECHO: 2000
```

MichaelAtOz, 2015-06-20:
Identifiers can start with numbers:

```
2r=3;
echo(2r);
```

```
Compiling design (CSG Tree generation)...
ECHO: 3
```

So not possible.

Steinhagelvoll, 2015-06-20:
Ok I have to agree this would just be a big mess. I believe to have seen it on wolframalpha, but they haven't got a full programming language.

FluxIX, 2015-06-28:
I'm also not convinced it would save all that much space on a line.

MichaelAtOz, 2015-06-28:
@Steinhagelvoll can this be closed?

Steinhagelvoll, 2015-06-28:
yes

closed 2015-06-28 by Steinhagelvoll