openscad/openscad #1373: Doesn't reload .scad file while animation in progress

darrenfreeman, 2015-06-23:
Running the package openscad-2015.03-2.1.x86_64 from the openSUSE build service. I think this is the most recent version, however it's not relevant to the following bug which has been there for a long time.

My usual workflow is to save the .scad file in a text editor, then look at the 3D view in OpenSCAD, which will have noticed the file update and automatically recompiled the scene.

With animation turned on, however, this doesn't happen. So now I have to save the file, go to OpenSCAD, uncheck "animate", check "animate".

I think it should probably detect the file change even during animation, thus this is a bug.

tcurdt, 2015-07-20:
This is becoming really annoying.
Could someone point me into the right direction inside the code to get this fixed?

clothbot, 2015-07-25:
Do you have "Automatic Reload and Preview" enabled, or even available as an option? That's what works on the Mac side, independent of whether animate is active.

Otherwise, "F4 - Reload and Preview" is how you have to manually trigger a reload of all dependencies.

tcurdt, 2015-07-25:
I am on Mac and that option is enabled. It does not work during a running animation though. I have to stop the animation and then it will reload automatically.

clothbot, 2015-07-25:
Are you using the latest development snapshot available here? http://www.openscad.org/downloads.html

OpenSCAD version 2015.06.21 (git 6bd23ab) 

That's what's working for me.

clothbot, 2015-07-25:
Here's what I've been testing it with:

https://github.com/clothbot/ClothBotCreations/tree/master/examples/variable_files

I filed #1400 against surface("surface.dat") file changes not being detected *except* when animations are running more for 2+ steps.

tcurdt, 2015-07-25:
I am on 2015.03-1 but also just tried 2015.06.21.

The dev version is even more broken. External saves are ignored as before, changes in the internal editor are reflected - but even on every keystroke (not just on save).

When I exit the program during a running animation it crashes with a SIGSEGV.

clothbot, 2015-07-25:
Yeah, I don't enable animation while actively editing with the internal editor...

You're not doing something like editing via the internal editor and an external editor in parallel, are you? I I wouldn't expect it to pick up changes to externally-edited file while you have unsaved internal-edits.

tcurdt, 2015-07-25:
Anyway - the last release works almost as expected.
Only the external save should also reload the animation.

Other than that - I am not using editors in parallel.
I am using an external editor - the internal one drives me mad :)

PetrBrabec, 2019-09-19:
Would be great to have the reload working while using external save. I have the same problem as @tcurdt. This issue is four years old, is there any plan to fix it?