openscad/openscad #1378: Linear extrude plus infinite twist causes crash instead of just an error.

christakahashi, 2015-06-30:
when I run the following:
    foo = 1/0;
    linear_extrude(height = 10, center = true, convexity = 10, twist = foo)
	square(3);

I get openSCAD to crash instead of giving an error message.

[image]

FluxIX, 2015-07-04:
Why ignore the parameter? This will make debugging difficult. Why not have a compile error (not program crash) instead?

t-paul, 2015-07-04:
It can't be a compile error, as the syntax is valid. It could be a run-time error and produce a warning. Right now the common behavior is to ignore invalid values.
I think changing this behavior should go into a specific issue (e.g. #596) and not as side-effect into a crash fix.

FluxIX, 2015-07-04:
I feel a run-time error is strongly preferential. NaN is not a value which should be ignored, as it indicates the value is invalid for an operation. Either infinity with an extrusion is the same argument.

Ignoring bad values and using a default instead is poor practice. Silently using defaults is very bad practice.

t-paul, 2015-07-04:
Yes, I agree. The patch addresses the crash issue only. That does not mean there are no other things that need improvement.

commit 1f5367bd (2015-08-12): Ignore Inf/NaN parameters to linear_extrude() (fixes #1378).
commit 2f4091b1 (2015-08-12): Merge pull request #1379 from openscad/issue1378-linear-extrude-crash

Ignore Inf/NaN parameters to linear_extrude() (fixes #1378).
commit 85aa71bb (2015-08-31): Ignore Inf/NaN parameters to linear_extrude() (fixes #1378).
closed 2015-08-12 by t-paul