openscad/openscad #1381: htho/scadlib

htho, 2015-07-07:
Hi,

I have created a library-tool for scad files. It also contains a specification proposal for documentation comments. It is a work-in-progress and does what I need. I bet it might be useful for someone else.

https://github.com/htho/scadlib

Please take a look at it and feel free comment and contribute.

Best Regards

htho

FluxIX, 2015-07-09:
1. I *really* do not like the Javadoc syntax. I prefer a syntax where the data format is very explicit as to what is being captured by the syntactic symbols, which the '@' tags and implicit description does not satisfy.
2. I do not feel the dependency tracking done here is sufficient. A human should not be responsible for determining what dependencies are present. The compiler already does this and the compiler does not forget to update its dependencies when you ask it to render. We should utilize the compiler to generate a dependency file for a `.scad` file. If the compiler is not desired for this task, a stand-alone script/program can do the task just as well, but the task task should be done by a machine.

htho, 2015-07-09:
Hi @FluxIX,

thanks for your comment.

**About the Syntax**
The big advantage of javadoc is its popularity. Using this syntax seems natural in an environment that uses a c-like syntax. (Everytime I use doxygen for C/C++, I am happy I already know javadoc)
I tried json (which would meet your requirements - I think?) and reStructuredText (which both already have python implementations) but the first was too much typing (xml would be worse) and the last was just unnatural.

**About dependency tracking**
You are totally right, keeping track of the dependencies should not be a users task.

Until now I avoided touching the content of modules, as parsing it seemed such a big task:
For example functions may be called from practically anywhere. Calls to modules and functions almost look the same, and there are local and global variables... Anyway, it is possible.

FluxIX, 2015-07-09:
**Syntax**
The XML documentation style Microsoft uses in the .NET development would be more typing if the IDE didn't give you the auto-completion. The other benefit of the format is you can place arbitrary XML tags in the documentation which can be processed (although no special use of the tag can be done), even though they weren't previously defined; this lets one extend the documentation syntax to accomplish new tasks.

I understand the annoyance with having to type more when documenting, but I feel the benefits do out-weigh the costs, in-part because the documentation is written every time you change what is being documented; the source item may be updated many times, but the documentation is only written once, then updated subsequently.

**Dependency Tracking**
I'd have to check the source code to determine how difficult it would be to extend the functionality of the parser to allow the automatic output of a dependency file. I also haven't worked out how to uniquely identify a symbol due to the lack of a namespace mechanism (#522).

htho, 2015-07-09:
**Syntax**
In general you are right, the [limited set of tags](https://msdn.microsoft.com/en-us/library/5ast78ax%28v=vs.140%29.aspx) is easy to memorize.

For key-value pairs this becomes tedious `<param name="name">description</param>` compared to `@param name description`
For my modules the documentation basically is a very large set of key-value pairs and lists (see [this Switch](https://github.com/htho/scadlib-electronic-components/blob/master/electromechanical/switches/switches/SPDT/ESP10XX.scad) as an example).

**Dependency Tracking**
Right  now the [constructor of ScadFileFromFile](https://github.com/htho/scadlib/blob/master/scadtoolLib.py#L974) holds the parsing logic, but this could be moved to a better place. There is some refactoring necessary anyway.

t-paul, 2015-08-09:
Thanks, maybe also post some info to the mailing list (http://www.openscad.org/community.html#mailing-list / http://forum.openscad.org).

closed 2015-08-09 by t-paul