openscad/openscad #1382: CGAL does not seem to acknowledge all polyhedrons

Zistack, 2015-07-09:
I've got a design for a hairstick that uses nothing but a for loop, arrays, and polyhedrons.  When I preview the model, it looks fine.  When I compile and render the model using CGAL, only the single polyhedron which is not in a for loop shows up.  When I export the model to STL, the STL also only has that single polyhedron in it.  There are 18 polyhedrons in all, and none of them are more complicated than a hexagonal prism.

This is what it looks like when I preview.
[image]

This is what it looks like when I compile and render.  The little piece visible is the polyhedron which is not in the for loop.
[image]

I have tried reordering the for loop and the lone polyhedron within the source file to no effect.  Putting the whole model into an explicit union also does nothing (not that I expected otherwise).

If I comment out the lone polyhedron, the model is empty when run through CGAL.  It seems that all of the geometry in the for loop is being ignored by CGAL.

t-paul, 2015-07-09:
Check if the orientation of all faces is correct and all polyhedrons are solid object without holes.
(See https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_is_my_model_showing_up_with_F5_but_not_F6.3F)

If that shows no problems, please add code to reproduce the issue.

Zistack, 2015-07-09:
Well, I had tried that before without any luck.  This time I saw the pink polygons right away.  Odd.

Thanks.

closed 2015-07-09 by Zistack