openscad/openscad #1383: Nested difference results in geometric CSG tree size increases

siberx, 2015-07-11:
While trying to create a model I ran into issues where my CSG tree would unexpectedly balloon up to huge sizes in certain circumstances; see the below test code for an example that produces over 5000 elements (and fails to render with a count any higher than 8). Each additional loop through for() doubles the size of the tree, quickly slowing the preview and eventually preventing correct preview rendering.

Performing a full render with F6 works fine, and results in a correct and responsive view. I'm running OpenSCAD 2015.03-1 64-bit on Windows 7 64-bit.

Note that the geometric expansion does not appear to occur if the inner difference() is the first element of the outer difference; it only shows up when the inner difference is placed as a second or later element. Additionally, the for() loop is not strictly required to demonstrate this; repeated inner difference operations will suffice as well but the test code is cleaner as shown below.

```
count = 8;
difference() {
	cube([50,50,20]);
	for(i = [0:count]) {
		difference() {
			cube([200,200,10]);
			cube([20,20,10]);
		}
	}
}
```

t-paul, 2015-07-11:
I'm not sure there is a general way to prevent the complexity explosion. For the preview that's a basic property of how it works. The actual display needs improvement though as the current OpenGL code used to render the graphics on screen is not using any GPU support provided by current graphics cards. Once this is available, it should be possible to raise the render limit quite a bit higher than currently possible.

That said, there is the workaround of using [`render()`](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Other_Language_Features#Render) to force mesh generation at this point (so the whole subtree below the `render()` statement is actually calculated by running the CSG calculations). This may slow down the first preview calculation, but the cache should help with the next previews as long as the tree below the `render()` does not change.

With `render()` at the inner `difference()`, even much higher values for `count` do not show problems (except the Z fighting display artifacts, but I assume those just the result of simplifying the example).
```
count = 100;
difference() {
    cube([50,50,20]);
    for(i = [0:count]) {
        render() difference() {
            cube([200,200,10]);
            cube([20,20,10]);
        }
    }
}
```

siberx, 2015-07-11:
You are correct that the Z fighting is simply a result of the naive example and doesn't exist in my actual scad (or a slightly tweaked example) although all produce the high element amplification shown here.

Applying a render() to the inner difference dramatically improves things; even the first preview generation is not significantly slower in my case, and the resulting tree contains 58 elements with my model's test case instead of a predicted 8 million or so if the scaling is happening the way I expect (it stops rendering correctly and clamps the element count with less than half the iterations I require).

Would it be possible to have OpenSCAD automatically detect this scenario and insert a render() as needed when generating the preview (ie: before inner differences())? From a previewing perspective, I can't see how a slightly longer (and cached!) render could be worse than simply failing to render correctly at all.