openscad/openscad #1384: multi expression functions

tcurdt, 2015-07-11:
The current support for functions seems very limited - or I don't get the syntax right.

Having a single expression in my case is just not feasible. It would make the whole thing super complex to maintain and it would calculate expensive sub results more than once. 

Something like this (pseudo code) example would be nice:

     function(a,b) {
       c = ... a ... b;
       d = ... a ... b;
       e = ... c ... d ...;
       return [ c,d,e ];
     }

MichaelAtOz, 2015-07-11:
Have a look at:
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/List_Comprehensions

t-paul, 2015-07-11:
The list comprehensions syntax is quite close to the suggested syntax. The only issue is that it's always producing a list of values so when returning a single vector the result needs to be unwrapped once (in the list comprehension examples this unwrapping is usually done by a separate `flatten()` function).

When  just always returning a single vector, it's easier to just select the first element of the list comprehension result using index access (`[0]`).

Example:
```
function f(a, b) = [ let (
    x = 30 * sin(2 * b),
    y = 60 * cos(b),
    z = (15 * a + b) / 10 )
    [x, y, z]
] [0];

for (a = [ 0 : 9 ], b = [ 0 : 5 : 715 ])
    translate(f(a, b))
        cube(1);
```

closed 2015-07-11 by t-paul

nophead, 2015-07-13:
I don't see why you would want to use a list comprehension and then flatten
it. This works: -

function f(a, b) =  let (
    x = 30 * sin(2 * b),
    y = 60 * cos(b),
    z = (15 * a + b) / 10 )    [x, y, z];

for (a = [ 0 : 9 ], b = [ 0 : 5 : 715 ])
    translate(f(a, b))
        cube(1);

t-paul, 2015-07-13:
Ah, right, good point. There's also the separate `let()` implementation. Totally forgot about that one :-).

https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Mathematical_Functions#let