openscad/openscad #1385: import() doesn't pick up the $fs value in use

darrenfreeman, 2015-07-13:
This may be DXF related, or a bug in the import() module interface.

These two code snippets produce different results, and I think this is a bug. I think it's been there for a long time. Note I haven't tested these two, I mean this as an example, and it could be slightly more complicated.

module good()
{
  $fs = 0.2;
  $fa = 0.01;
  rotate_extrude() import(file = "foo.dxf", $fs = 0.2);
}

module bad()
{
  $fs = 0.2;
  $fa = 0.01;
  rotate_extrude() import(file = "foo.dxf");
}

t-paul, 2015-07-13:
I can't see any difference using a DXF (exported from LibreCAD) with just a circle.

[image]

Can you please add a full example that shows the problem?

What version are you using?

MichaelAtOz, 2015-07-13:
In good() I would think the $fs as a parameter should go into the rotate_extrude() not the import()?
How does $fX apply to import, it should generate the DXF as defined,not modify it?

darrenfreeman, 2015-07-14:
Michael, you can put $fs into either the rotate_extrude() or the import(), and they control different things. Imagine turning a circle into a torus, using rotate_extrude($fs = x) circle($fs = y). The x and y parameters control the mesh spacing in orthogonal directions. The same thing happens if you put a circle into a DXF and then import($fs = y).

t-paul, of course you can't see a difference, your object is huge compared to $fs. Use a tiny circle, r = 1, in your DXF.

I am using OpenSCAD 2015.03 as supplied by openSUSE in openscad-2015.03-2.2.x86_64.

darrenfreeman, 2015-07-14:
Also t-paul you are using $fs = 1. Check whether this is the default used by the import() call, as that would explain why nothing changes. Try an object that needs a much smaller $fs.

darrenfreeman, 2015-07-14:
In fact why did you change my values? I use $fa = 0.01 so as to force it to go with the $fs setting. If it's $fa limited, and if the bug doesn't apply to $fa, then that's another reason you can't reproduce the bug.

Please try again without changing my code.

t-paul, 2015-07-14:
Oops, I made a screenshot from the wrong test then. I tried to change the values to actually get a difference. But that did not help at all. I'll try with r = 1...

Are we talking about #999 maybe?

darrenfreeman, 2015-07-14:
I don't think it's #999.

I have had issues where I had to scale up text in my DXF by 10x and then scale it down again in OpenSCAD, or otherwise curves would turn into lines. But I think in hindsight the problem was that the $fn value wasn't being taken from the context of the function call. Since I've started always giving $fs in an import() this hasn't been happening.

t-paul, 2015-07-15:
I still can't see any difference. It might depend on the entities used in the DXF.

Please add a full example that shows the issue.

[image]

darrenfreeman, 2015-07-16:
Okay, I've worked it out. You need an additional redefinition of $fs in the rotate_extrude. I've tested the following and it reproduces the bug. Without the extra redefinition it doesn't reproduce, so I don't think it has anything to do with my DXF. Try:

    module good()
    {
      $fs = 0.5;
      $fa = 0.01;

      rotate_extrude($fs = 5)
      {
        import(file = "tmp.dxf", layer = "rotate-extrude", $fs = 0.5);
      }
    }

    module bad()
    {
      $fs = 0.5;
      $fa = 0.01;

      rotate_extrude($fs = 5)
      {
        import(file = "tmp.dxf", layer = "rotate-extrude");
      }
    }

    translate([0, 0, 10])  good();
    translate([0, 0, -10]) bad();

darrenfreeman, 2015-07-16:
I just want to confirm that this is a bug and not expected behaviour. I certainly found it surprising that putting $fs in the rotate_extrude() should be then inherited by its children. I don't think I've seen this before.

darrenfreeman, 2015-07-16:
It still happens if you replace the import() above, with:

    translate([20, 0]) circle(r = 1, $fs = 0.5);

So I guess this is intended behaviour. Since I've never used my own special variables, I had no idea this was happening!

https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Other_Language_Features#Special_variables

closed 2016-01-08 by t-paul