openscad/openscad #1390: Transparency is dependant on order of objects.

jonathanimb, 2015-07-16:
Objects with transparency should be rendered last, but objects are rendered in the order that they are written in the program. Example code to illustrate the problem:

	module works() {
		cylinder(r=1, h=10, center=true);
		
		color("blue", 0.5)
		translate([3,0,0])
		cube([1,10,10],true);
	}

	module no_works() {
		color("blue", 0.5)
		translate([3,0,0])
		cube([1,10,10],true);
		
		cylinder(r=1, h=10, center=true);
	}

	translate([0,-8,0])
	works();

	translate([0,8,0])
	no_works();

	$fn = 100;

[image]

Thanks for all your work, please keep it up!

alamar, 2015-09-26:
Moreover, I would prefer much that objects will be co-transparent: for two transparent objects you can see the first thru the second and then, when you drag it around, second thru the first.

ygoe, 2022-05-13:
Not sure what "co-transparent" means but the transparency seems useless to me as it is now. It's just unreliable and doesn't work most of the time. Too frustrating to even bother. A profile with deeper parts to the inside won't even let its next surface show through. This is something that cannot be fixed by the order of objects, it's the same object.

jordanbrown0, 2022-05-14:
Here's what he means by "co-transparent".

Given:
```
translate([0,4,0]) color("green", alpha=0.3) cube([10,1,10]);
color("red", alpha=0.3) cube([10,1,10]);
```
if you look at it from -Y, you see:
[image]
but if you look at it from +Y, you see:
[image]

For some reason it's more visible on my screen than in these captures, but note that in the first the green is at least somewhat visible behind the red, but in the second the red is not at all visible behind the green.

ygoe, 2022-05-14:
Ah you mean transparent in both directions. I mean transparent to itself. Neither is true at the moment. Both are needed for colour transparency to be frustration-free.

But as I'm making screenshots of my model, I see that the self-transparency might in fact be restricted by the same ordering issues described here. My profile is a linear extrusion of a polygon, and the order of the points in that polygon seems to be relevant for this as well.

Here's the view from the top:

[image]

And here from the bottom: 

[image]

However I turn it, the front-facing parts of the upper side profile are not visible.