openscad/openscad #1391: Feature Request: Could we get an icon for View/Diagonal in the preview toolbar

MichaelAtOz, 2015-07-18:
Please.

kintel, 2015-08-18:
I think the View icons should be collapsed into some sort of combined icon, as they take up a lot of screen real-estate. Once that happens, this should be trivial.

MichaelAtOz, 2017-05-16:
Suggest the icon could be a small cube wire-frame (like the existing ones) without arrows or solid face, oriented with dotted axis lines to the edge of the icon. Notionally like so:
[image]

kintel, 2017-05-16:
Right now this is pretty much what "Reset View" does.
As mentioned earlier, I'm hesitant to add more icons without also collecting some of the existing icons into e.g. icon "combo boxes"

MichaelAtOz, 2017-05-16:
> they take up a lot of screen real-estate.

I presume on a laptop? On a 1920x1080 (fairly low res now) they are trivial.
On my NEW MONITOR 3440x1440 they are tiny....(had to have a better monitor for my 8 core)

> View icons should be collapsed into some sort of combined icon

Having a collapsible icon list is sort of past the point of icons, its just a menu with icons instead of text, may as well use the view menu, same two clicks.
If you could select icons you'd like that would be good, but I imagine it is non trivial.

> Right now this is pretty much what "Reset View" does.

Reset View changes the size and repositions the view, annoying if the object is irregular, to switch back to the other views you have to resize/reposition. Anyway...

I got an overwhelming (0) response to my query in the [forum](http://forum.openscad.org/Thought-bubble-Preview-icons-vertical-or-undockable-td21065.html):

> I though about other apps, many have vertical icons on the edge of the window.
> I presume most people have the window taller than wider, so there would be more room. (?)
> 
> Another possibility is to make it undockable, then it could be sized independently.
> 
> Discuss... 

Or have both a horizontal & vertical icon bar?

Anyway, I was just going thru my Issues list, saw this one thought about the icon design and posted my thoughts. I wasn't bumping it. (much)

closed 2026-02-05 by github-actions[bot]