openscad/openscad #1392: Bug? if not, Feature Request: if {} else if {} code collapse in editor

MichaelAtOz, 2015-07-18:
See below.

The red highlight shows that the `if `and `else` parts can be collapsed separately.
The blue, using `else if` cannot, it collapses the whole statement. Not so handy if it has many `if else`'s and is 4 pages long.

[image]

(code for cut/paste)
```
// if & else if

if (true)
	cube(1);
else
	cube(2);

if (false) {
	cube(3);
}

// syntax break
if (true)
	{
	cube(4);
	cube(5);
	}
else
	{
	cube(6);

	}

// syntax break
if (true) {
	cube(7);
	//
} else if {
	cube(8);
	//
} else if {
	cube(9);
	//
} else if {
	cube(10);
	//
}

// syntax break
```

MichaelAtOz, 2015-07-20:
This is what it looks like in Notepad++

[image]

kintel, 2016-08-20:
Fixed by #1776

closed 2016-08-20 by kintel