openscad/openscad #1393: color overrides transparency

tcurdt, 2015-07-21:
Here is an example:

    module test()
    {
      % cube([10,10,10], center=true);
    }

    color("blue") 
    translate([0,-10,0]) test();
    translate([0,+10,0]) test();

The module defines the cube as transparent - yet the color is applied to all childs also overriding the transparency.

It probably is debatable if this is what should happen - but would expect the blue cube to be blue and transparent.

MichaelAtOz, 2015-07-21:
See issue #492 is prob covered there. See also #366  #310 and #304.

Please close if you think it is a dup.

bobc, 2015-07-22:
Color (rgb) and transparency (alpha channel) could be considered as separate properties, but normally alpha channel is a meta-property defining how colors are blended.

In the general case, I would always expect properties to be applied down the tree, not up.

In this case, the % is a special operator meaning "hidden". In the preview, hidden objects are displayed as transparent gray, but in render view are not displayed at all. Therefore, I would expect the % modifier to be "sticky" and never be overridden.

tcurdt, 2015-07-22:
I think the behaviour @bobc describes is probably the one with the least headaches.
So +1 for that.

While #492 is related this isn't quite the same. #304 comes very close - but is not quite the same either.
Plus it is closed. So I am not quite sure this issue qualifies as a dup.

MichaelAtOz, 2015-07-22:
> In this case, the % is a special operator meaning "hidden". In the preview, hidden objects are displayed as transparent gray, but in render view are not displayed at all. Therefore, I would expect the % modifier to be "sticky" and never be overridden.
>
>Therefore I would expect both cubes to appear as transparent gray, as both are eliminated in render view.

I went to some trouble (#304 etc) to get colour override into the % modifier. Specifically for preview to facilitate visual documentation/context with color and transparency. As long as that is not lost. However that is leading to some perceived inconsistency, such as the above. If the inconsistencies can be improved, they need to NOT go back to grey all over.

I did argue at the time that the % modifiers problem was that it did two things, it kept items from the final render, and it had visual effects. Inevitably such behavior comes to grief when someone wants one function without the other.

At the moment I believe the behavior is, don't fiddle with colour already applied, but use grey 50% if no color() is specified, AND don't render. So the above code is consistent with this.

tcurdt, 2015-07-23:
@MichaelAtOz Turning just gray with 0.5 transparency would be easy to understand.
Hence my +1 for @bobc. 

That said - I would prefer if % would just use the current color (which could also be the default color) and either only set the transparency component to either fix 0.5 or 0.5 of the current color transparency.

I guess that would get my +10 then ;)

The way it is right now really is really inconvenient for debugging.

MichaelAtOz, 2015-07-23:
And that is the problem. Now 3 functions. Stop Rendering, affect colour, used for debugging.

I have also argued previously for a proper debugging mechanism.

I suspect 50% of the current transparency (and not changing a pre-specified colout), and if no colour/transparency is specified, go grey 50%.

However, the children() and module() "scope" needs to be carefully considered.

I would like to be able to do:
```
module makeBlue() color("blue") children();

module debug(d=0) 
    if (d=0) children();
    else colour(  ( (d==1) ? "yellow" : "red" ), ( (d==1) ? 1  : 0.25) )
        if (d==1) %children();
        else children();

debug(1) makeBlue() cube();
// and have an understanding of what is/should be the colour, transparency and rendering effect.
```
Unfortunately, there is no formal specification of the language. So that is totally unpredictable.

tcurdt, 2015-07-23:
I would love to understand your usecase - but you lost me with that code

    d==0 children()
    d==1 color("yellow", 1) %children()
    d==else color("red", 0.25) children()

why the yellow/red transparent/not transparent and render/not render thing?

MichaelAtOz, 2015-07-23:
Firstly it was just an example, I have other cases I wanted to do similar, but didn't work.

With `debug() yourObject();` it works as normal. (I would probably change debug() to d() to save typing)
You can change that to `debug(1) yourObject();` it will be shown yellow solid, not rendered, so if you do F5 it shows, F6 it doesn't - you can see all objects around it.
And `debug(2) yourObject();` transparent red. You could go on to define d=3 etc for other effects.

tcurdt, 2015-07-24:
Frankly I don't really see the benefits of `d(1)`.
I am wondering: Why would you want it as solid in the preview but then not rendered?
Isn't that a super corner case where to build on side effects?

Maybe `%` should really only mean "remove from rendering".
The default would make it always gray - not matter what.
But would also let us define the color (modification) of such a case.

tcurdt, 2015-07-24:
Ideally it would be a function - but for me a changeable default color would be good enough.
This could be the default implementation:

    $unrendered_color = function(original_color) {
      return color("gray", 0.5);
    }

Not sure how that internally would work with nested nodes though.

    % union() {
      % cube()
    }

tcurdt, 2023-01-20:
I still haven't found a workaround for this.

This time it's the other way around.
Why are all the red cubes solid in the last row?

```
module obj() {
  cube([10,10,10], center=true);
}

d = 20;

 %translate([0*d,0*d,0]) obj();
 #translate([1*d,0*d,0]) obj();
#%translate([2*d,0*d,0]) obj();
%#translate([3*d,0*d,0]) obj();

 %translate([0*d,1*d,0]) color("red",0.5) obj();
 #translate([1*d,1*d,0]) color("red",0.5) obj();
#%translate([2*d,1*d,0]) color("red",0.5) obj();
%#translate([3*d,1*d,0]) color("red",0.5) obj();

 %translate([0*d,2*d,0]) color("red") obj();
 #translate([1*d,2*d,0]) color("red") obj();
#%translate([2*d,2*d,0]) color("red") obj();
%#translate([3*d,2*d,0]) color("red") obj();
```

[image]